keyboardio/Chrysalis #548: Oneshot modifiers menu missing "RIGHT SHIFT" in Chrysalis 0.7.9

tmarzolf, 2020-08-07:
**Describe the bug**
Right Shift should be included in the "OneShot modifiers" menu

**Desktop (please complete the following information):**
 - OS: Windows
 - Chrysalis Version: [e.g. 0.7.9]

**Screenshots**

[image]

**Additional context**
 This appears to have existed in previous versions as I have "OSM RShift" configured even though the option no longer exists

algernon, 2020-10-09:
Hm, this is interesting... the key _is_ in the database, so something somewhere filters it out, it appears.

algernon, 2020-10-09:
It seems that we only ever display 5 keys in that group... if I remove `Right control` from the OSM group, `Right Shift` (the next in the list) will show up. This is one funky bug.

algernon, 2020-10-09:
HAH. Found it. When we wanted to restrict OSL keys to show only layers we have available, I typo'd the group, and we restricted the number of OSM keys to the number of available layers instead.

commit 2bafb1e8 (2020-10-09): KeySelector: Update the groups that contain layer keys

We want to limit groups that contain layer keys to only show as many keys as we
have layers available. With the introduction of `MoveToLayer`, our groups
shifted, and we ended up putting OneShot Modifiers into this group, instead of
OSL, and completely missed `MoveToLayer`.

Updated the check, so it applies to the correct groups again.

Fixes #548.
closed 2020-10-09 by algernon