keyboardio/Chrysalis #551: Model 01: JS error after installing latest firmware - white screen

thejonanshow, 2020-08-12:
**Describe the bug**
1. Install firmware
2. Connect
3. Admire JS errors in console

**Expected behavior**
None errors instead of some.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Chrysalis Version: 0.7.9+258

**Additional context**
Tried installing firmware from Chrysalis (beta) and then reinstalled with firmware from Arduino, same result.

obra, 2020-08-12:
Hi Jonan,

Do you also get this error on Kaleidoscope 0.7.9, the latest release?

thejonanshow, 2020-08-12:
I used the Arduino examples Model 01 firmware most recently, can I assume by your question that this is not the latest version?

obra, 2020-08-12:
So sorry. I meant to write "Chrysalis 0.7.9" - the latest 'release' build

algernon, 2020-10-09:
This has been fixed on master, and the fix will be available in the next release.

closed 2020-10-09 by algernon