keyboardio/Chrysalis #557: Cannot permanently set 'Default Layer'

clemensw, 2020-09-10:
**Describe the bug**
I was confused why my customizations were not working until I saw 'hardcoded layers'.
Setting Default layer to 'Layer 3' activates my changes, but does not persist across power-cycles.

**To Reproduce**
Steps to reproduce the behavior:

1. Make a customization to Layer 3
2. Save changes
3. Select 'Preferences'
4. 
  - Observe -> 'Show hardcoded layers': off
  - Observe -> 'Use custom layers only': off
  - Set 'Default layer' to 'Layer 3'
5. Save Changes

**Expected behavior**
- 'Default layer' display in 'Preferences' remains 'Layer 3'
- Customization is active when using keyboard
- Customization remains active when power-cycling keyboard

**Actual behavior**
- After green 'Changes Saved success' flash 'Default layer' reverts to 'No default'
- Customization is active when using keyboard (though I haven't yet seen my custom colormap)
- Customization is no longer active when power-cycling keyboard

**Screenshots**
If applicable, add screenshots to help explain your problem. If possible, please also include a log from the "developer console". You can access it by pressing `Control + Shift + I`. When it appears, right click inside it, and select "Save as", and attach the saved file to this report.

**Desktop (please complete the following information):**
 - OS: MacOS 10.15.6
 - Chrysalis Version: 0.7.9 (brew)
 - HW: Keyboardio Model01 flashed by Chrysalis to 'Chrysalis 0.7.9 experimental'

**Additional context**
This can probably be worked around by setting 'Use custom layers only', but this has so far led to more confusion:
- Custom layers are downshifted which makes layer changing keys point to the wrong layers
- The colormap for a layer is not downshifted, but remains orphaned on an otherwise empty layer
- While the initial display seems about right restarting Chrysalis shows empty keymaps similar to #519

rca, 2020-11-25:
I can confirm that the preference pane in Chrysalis does not display that a non-default layer has been set.  However, unplugging and re-plugging my keyboard does retain the default layer to be Layer 3 instead of 0.

clemensw, 2021-01-12:
Thanks for confirming.  I wasn't able to get the default layer to stick until reflashing with "Restore to factory defaults". Now the default layer is kept, but (as you say) not displayed.

algernon, 2022-05-17:
We removed the set of options (showing hardcoded layers, etc), and Chrysalis now requires that the keyboard use EEPROM-layers only. This, in turn, makes setting the default layer reliable. As such, I'm closing this issue.

closed 2022-05-17 by algernon