keyboardio/Chrysalis #558: Sending multiple keys for typing international characters

faxm0dem, 2020-09-18:
Hi,

My main use-case here is to be able to type international chars in an US layout.
The way this could be achieved with os-support would be to send multiple keys subsequently,
for example to type `é` it could send `compose`, `'`, `e`.

Rom1deTroyes, 2020-09-18:
On my GNU/Linux computers, it's working using US-Intl layout : the right alt key becomes an <kbd>alt-gr</kbd>, and works like a <kbd>Compose</kbd> key on my Sun keyboard.

```bash
xinput -list
⎡ Virtual core pointer                    	id=2	[master pointer  (3)]
⎜   ↳ Virtual core XTEST pointer              	id=4	[slave  pointer  (2)]
⎜   ↳ USB Optical Mouse                       	id=15	[slave  pointer  (2)]
⎜   ↳ Keyboardio Model 01 Mouse               	id=10	[slave  pointer  (2)]
⎜   ↳ Keyboardio Model 01 Consumer Control    	id=13	[slave  pointer  (2)]
⎜   ↳ Keyboardio Model 01                     	id=20	[slave  pointer  (2)]
⎣ Virtual core keyboard                   	id=3	[master keyboard (2)]
    ↳ Virtual core XTEST keyboard             	id=5	[slave  keyboard (3)]
    ↳ Power Button                            	id=6	[slave  keyboard (3)]
    ↳ Power Button                            	id=7	[slave  keyboard (3)]
    ↳ HID 0430:0005                           	id=16	[slave  keyboard (3)]
    ↳ Keyboardio Model 01 Keyboard            	id=9	[slave  keyboard (3)]
    ↳ Keyboardio Model 01                     	id=11	[slave  keyboard (3)]
    ↳ Keyboardio Model 01 System Control      	id=12	[slave  keyboard (3)]
    ↳ Keyboardio Model 01 Consumer Control    	id=14	[slave  keyboard (3)]
    ↳ UVC Camera (046d:0825)                  	id=8	[slave  keyboard (3)]

setxkbmap -device 9 -layout "us,fr" -variant "altgr-intl,bepo" -option "grp:alt_shift_toggle"
```
(here I can change from BÉPO to US-International with Alt+Shift on my Model01, living the other keyboards as default)

faxm0dem, 2020-09-18:
I just discovered the existence of `altgr-intl`, thanks!
I however fail to see how I can generate a `è` or `à` for instance

Rom1deTroyes, 2020-09-18:
I found [this article](https://choam.eu/altgr-intl/) very instructive about all this...

Basically, in french, I use :

- <kbd>AltGr</kbd>+<kbd>6</kbd> for `âêîôû` (^, but not shifted)
- <kbd>AltGr</kbd>+<kbd>'</kbd> for `é` (the key with `"`)
- <kbd>AltGr</kbd>+<kbd>Shift</kbd>+<kbd>'</kbd> for `ë` (so a `"` on the `e`)
- <kbd>AltGr</kbd>+<kbd>\`</kbd> for `àèù` (same key as `~`, hard to write in markdown :cry:)

faxm0dem, 2020-09-23:
Thanks so much. I read the excellent article, and currently I'm trying out the `altgr-weur` layout.
Looks very promising!