keyboardio/Chrysalis #559: Discard prompt still shows "cancel" and "OK"

QEDeD, 2020-09-30:
**Describe the bug**
Discard prompt button naming appears to be inconsistent with the changelog. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Layout & colormap editor
2. Click on a button and try to edit the color
3. Click on the X in the top left corner 
4. See Discard Prompt that shows up

**Expected behavior**

> Chrysalis 0.7.6
> Clearer message when discarding changes
> 
> When leaving a screen with unsaved changes, the button is now labeled "Discard" instead of "Cancel", to make it clearer what it actually does: discard the changes, not cancel the operation.

The buttons do not seem to be labeled as they should, based on the changelog as quoted above. Further, my own thoughts are that it would make sense if the options were "cancel" and "discard". 

Cancel = stop the current operation/action of leaving the screen and the second option being changed from "OK" to "Discard" to emphasize that you are Discarding your changes.  

**Screenshots**
[image]

**Desktop:**
 - OS: Windows 10
 - Chrysalis Version: Chrysalis 0.7.9

algernon, 2020-10-09:
It turns out that the NEWS entry is wrong. We changed the _title_ of the dialog, not the button label.

In the context of the title and the description, I think `Cancel` and `OK` are the correct labels for the buttons. `Cancel` does cancel the discard operation, and `Ok` says "yeah, go ahead with discarding the changes".

I'll retroactively update the NEWS, as I believe that the current shape of the dialog is fine.

QEDeD, 2020-10-12:
I agree 👍

closed 2020-10-12 by QEDeD