keyboardio/Chrysalis #560: Insufficient permissions, please make sure the device file is read- and writable on Ubuntu 20.04

bolob, 2020-09-30:
**Describe the bug**
Launch Chrysalis (Appimage 0.7.9+259), Error message pops up.

Did also happen after I set myself in the dialout group, logout/login

**To Reproduce**
see above

**Expected behavior**
No Error message

**Screenshots**

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04
 - Chrysalis Version: [e.g. 0.4.0]

**Additional context**
Add any other context about the problem here.
[image]

Solution found with help on discord (algernon) :

Download this file: https://raw.githubusercontent.com/keyboardio/Kaleidoscope/master/etc/60-kaleidoscope.rules, place it in /etc/udev/rules.d/ (using the same name, 60-kaleidoscope.rules), then sudo udevadm control -R, and then if you logout and back in, it should hopefully work.
what that file does, is tell udev & systemd to give ownership to the atreus (& model01) device files to the user logged in on the main "seat" (which in practice translates to whoever is logged in on the desktop)
might even work if you just unplug & replug the keyboard, as udev should re-detect it then, so you can save a logout & login

jthegedus, 2020-10-03:
1. `sudo curl https://raw.githubusercontent.com/keyboardio/Kaleidoscope/master/etc/60-kaleidoscope.rules > /etc/udev/rules.d/60-kaleidoscope.rules`
2. `sudo udevadm control -R`
3. logout
4. login
5. disconnect, reconnect & scan for devices

algernon, 2020-10-03:
FWIW, you don't even need to logout and back in - if you unplug & replug the keyboard, that should be enough for udev to notice, and set up permissions appropriately. `sudo udevadm trigger -v /dev/ttyACM0` (or whatever the port ends up as) should also work, but I need to double-check to be certain.

xerus2000, 2020-10-04:
I have the script in there, can see it is working (symlink `/dev/model01` exists and is correct), still getting this error :/

algernon, 2020-10-04:
Are you using Gnome or KDE, or anything else that plays well with systemd? The udev script relies on systemd knowing who is logged in, and that needs a bit of support from the login manager / DE.

xerus2000, 2020-10-06:
I use KDE Neon on Ubuntu 20 LTS, which ships the latest KDE Plasma :)

algernon, 2020-10-06:
@xerus2000 Can you tell me what `loginctl list-seats --no-pager` displays?

Ideally, it should be something like this:

```
❯ loginctl list-seats --no-pager
SEAT 
seat0

1 seats listed.
```

If it doesn't list a seat, then the udev rules as they are now will not be enough, and we'll have to figure out something else.

GrantCuster, 2020-10-07:
I had this issue with my new keyboardio Atreus and fixed it by adding the rules suggested above (on Ubuntu 20.04).

I also have an older Atreus keyboard and I'd love to run it through Chrysalis as well, so that I can share layouts between them. But I get the insufficient permissions error when selecting the original Atreus. Is there a different rule I need to set for the original Atreus?

[image]

lee2sman, 2020-10-07:
Hi, I just received my Atreus today and had the same problem and here is an easier alternate solution! I jumped onto discord and someone suggested I needed to add permissions for my user to the ```dialout``` group. So I did, and it worked a charm. Didn't need to download anything else or change any files.

So just run in the Terminal:

```
sudo adduser $USER dialout
``` 

And then log out and back in. Then re-open Chrysalis. Voila. 

I'm on Ubuntu 18.04 by the way.

skyfaller, 2020-10-07:
`sudo adduser $USER dialout` worked for me after a reboot on Pop!_OS 20.04 LTS focal using Wayland. Thanks @lee2sman !

bolob, 2020-10-07:
cool, sudo adduser $USER dialout   didn’t wok for me after logout/login (but I was told that reboot wasn’t needed).

algernon, 2020-10-07:
The `dialout` group only works if the device file is owned by `root:dialout`, and is group writeable. Different distributions use different groups for the file, so it might not be `dialout` at all. The easiest way is to install the udev rules, which will work on any systemd-based system.

The master branch now has a better explanation too, and a "Fix it" button (which installs the udev rules, reloads it, triggers an udev rescan for the device, and then checks if it is accessible after).

algernon, 2020-10-09:
I think that the new changes on master adequately fix the issue for the vast majority of cases, so I'm going to close this one. Do feel free to reopen (or open a new issue) if Chrysalis master does not address the issue on some distro or another.

closed 2020-10-09 by algernon

xerus2000, 2020-10-13:
Btw, seats output:
```sh
❯ loginctl list-seats --no-pager
SEAT 
seat0

1 seats listed.
```

xerus2000, 2020-10-13:
The udev rules are in place and working, but Chrysalis only worked with the above dialout tweak...