keyboardio/Chrysalis #562: Support for importing / exporting complete layout

pm0u, 2020-10-02:
**Is your feature request related to a problem? Please describe.**
There should be a way to export and import complete keyboard layouts (vs single layer at a time)

**Describe the solution you'd like**
A UI based option on the keyboard layout edit screen that allows pasting or otherwise importing (perhaps a JSON file?) and entire keyboard layout

**Describe alternatives you've considered**
None

obra, 2020-10-02:
We need a way to export/import the full keyboard configuration, including all layers and LED configuration. It should -also- include metadata about what keyboard it was sourced from. 

There are two eventual use cases I'm envisioning:

1) Backup / Restore
In this case,  we -should- include all the metadata from EEPROM, I think.

2) Sharing full keyboard configs

In this case, when exporting, I don't think we want to include any opaque non-layout/LED eeprom data, as it might contain macros.

For an initial implementation, one big text area with ugly-looking JSON is acceptable. My only requirement on the json structure that the json have a 'version': key, so that we can untangle it in the future.

kajsa, 2020-10-06:
my work-around for my own uses was to create a gist, with each layer as a "file" layer<n>.json.

algernon, 2021-01-25:
I believe this is mostly, if not fully, covered by 0.8.0?

closed 2024-02-26 by obra