keyboardio/Chrysalis #564: Gui modifier key checkbox should be localized

alfalfasprossen, 2020-10-02:
The *Gui* modifier checkbox for combining with other keys is not localized in a system-specific way. It is correctly translated on the key labels and in the *Modifiers and Locks* category. The checkbox should be localized as well to prevent confusion.

[image]

**Desktop (please complete the following information):**
 - OS: Windows 10, MacOS
 - Chrysalis Version: 0.7.9

Rom1deTroyes, 2020-10-07:
Does the TODO here is to extract the substring "Gui" from all the place in `src/api/keymap/db/*.js` , and replace with the `guiLabel` var from `src/api/keymap/db/gui.js` (like in the OSM file) ?

Is there a plan for extracting other things (I think for internationalization : Right / Left, Enter, Escape, ...) ?

algernon, 2020-10-07:
I wouldn't worry about this just yet - I'm rebuilding both the database and the key selector to work _very_ differently. We'll need to do the localisations (both language & platform ones) very differently.

algernon, 2021-01-25:
This is partially fixed on master - the Gui key is localized both on the keymap, and in the sidebar. The short label is not localized yet, though. I'll see if I can fix that easily.

commit 1fd05c73 (2021-01-25): api/keymap: Localize the short label of Gui keys

When displaying the Gui key's label shortened, use a platform-specific
shortening.

This, along with the editor & db rework in 0.8.0 fixes #564.
closed 2021-01-25 by algernon