keyboardio/Chrysalis #566: one-shot-modifier does not work if next key is a modifier-on-hold

talw, 2020-10-04:
**Describe the bug**
One-shot-modifier does not work when the key to modify has a setting where on hold it acts as a modifier key

**To Reproduce**
Steps to reproduce the behavior:
1. set a key with modifier on hold. In my case `s` is winkey on hold.
2. set a key one-shot-modifier, in my case shift

**Expected behavior**
After saving the configuration. Press the one-shot-modifier shift key, next press `s`, an upper-case s does not get input (which is why in this bug report all my `s` are lowercase.

If trying to press a different letter without a modifier-on-hold like `a` after the one-shot-modifier shift key, it works as expected: `A`.

**Desktop (please complete the following information):**
 - OS: Windows for Chrysalis
 - Chrysalis Version: 0.7.9

talw, 2020-10-10:
Since QMK has merged keyboardio/atreus support to master, I switched to QMK so I am not waiting for this to be resolved.
Since this is a bug I assume other users are interested in this being resolved, and so I am not closing the issue.

gedankenexperimenter, 2021-06-02:
I believe this is the same problem as #605, and both work properly in Kaleidoscope now.

algernon, 2022-05-17:
This should be fixed by an updated firmware, so closing it. If it is still a problem, please reopen.

closed 2022-05-17 by algernon