keyboardio/Chrysalis #567: double press and hold on second press of layer-on-hold key should repeatedly output the original key instead of switching layers

talw, 2020-10-04:
**Is your feature request related to a problem? Please describe.**
I use QMK for my other keyboards and my configuration uses the backspace key as a layer-switch-on-hold.
Whenever I want to repeatedly output backspace I double press the key and leaving the key held down on the second press.
That way the firmware knows it's not the layer-switch but a repeat of the original key (the backspace in this case).
I also have other keys besides backspace with the same problem (arrow-keys for example), so at the moment I can not use the keyboard until QMK adds support or Chrysalis adds this feature (that is, unless I reconfigure a large part of my  keymap to accommodate for the lack of this functionality)

**Describe the solution you'd like**
Ideally, I would like this to have a similar functionality to QMK where if a key is tapped once and then held down in quick succession it does not switch layers but instead repeatedly output the original key.

**Describe alternatives you've considered**
Completely revising my 5 years old layout as I shuffle keys around.

Any help is greatly appreciated!

talw, 2020-10-10:
Since QMK has merged keyboardio/atreus support to master, I switched to moving QMK so I am not waiting for this to be implemented.
Should I leave this issue open for other users who are not using QMK and are interested in the same feature?

obra, 2020-10-10:
Please do, thanks.

gedankenexperimenter, 2021-06-02:
This feature is now available in Kaleidoscope (called Qukeys tap-repeat).  I'm not sure, but I think it has one minor difference from the QMK Mod-Tap version, in that the tap & hold sequence in Kaleidoscope Qukeys will result in just a single press and hold event being sent to the host.