keyboardio/Chrysalis #572: We should display the version number on the screen, possibly just in the menu bar as Chrysalis v0.88+123 instead of Chrysalis

obra, 2020-10-07:
**Is your feature request related to a problem? Please describe.**

What version of Chrysalis has this bug?

[image]

algernon, 2020-10-07:
We already display the version when the main menu is opened (below the menu items):

[image]

It will show the latest release for local and release builds, but will show the +N for builds via Travis. I can change the window title, but... that's a bit awkward to do, and looking at any of my open windows, none of them have the version in the window title. (Emacs does, by default, but I don't think we want to copy Emacs when it comes to UI decisions.)

obra, 2020-10-07:
I'm ok with it not being in the window title. But I'm definitely running
into support issues trying to reverse engineer the actual running version
from screenshots.

algernon, 2020-10-07:
I guess the title it is, if asking for a screenshot with an open menu isn't a practical option. Because other than the title and the opened menu, I don't know where else I could put the version. Perhaps on the footer, but... that looks so awkward, and eats precious screen space.

obra, 2020-10-07:
Opening the menu is going to obscure information. Let's leave this open for
now and come back to it.

algernon, 2020-10-07:
I have a couple of ideas how to get better info out of Chrysalis, along with a screenshot - because the goal is to have as much info as possible, and a screenshot of the current screen, as easily as possible, right?

We could add a key binding that would take a screenshot (Chrome can do that for us), and bundle a bunch of other info (OS, Chrysalis version, the dev console contents, etc), pack that into a zip, and pop a save dialog. Then customers can just send you the zip, and you have all the info.

obra, 2020-10-07:
Yeah, that'd be really good.....

algernon, 2020-10-09:
Do we still need this with #575 merged?

obra, 2020-10-09:
nah

closed 2020-10-09 by obra