keyboardio/Chrysalis #580: Layer reordering

shader, 2020-10-10:
It would be awesome if there was a way to reorder the layers, such as buttons for moving the current layer up and down, instead of having to export/import them, or temporarily copying them to another layer.

My use case is that I wanted to add a second main layout (qwerty, since I normally use colemak), but realized that if I want to be able to shift to layers 1 and 2 I need to insert it below them.

Or is that already a feature and I just didn't see it?

algernon, 2022-06-02:
We have recently added drag and drop support to the Macro Editor, and that reminded me of this feature request... we could let the user drag the layers on the overview in the sidebar, and rearrange layers that way. We'd add a draggable handle to the table there, and let the dragging begin.

That'd make it very easy to switch layers around.

Rom1deTroyes, 2022-06-03:
It will be a very useful feature !

I have 2 use cases around this :
1. Re-odering my layers (I want to delete some unused layers, and move one on the top of others)
2. Having a draft layer and replace an existing layer when it is ready

On the second one, all the move and switch to layers of the other layers must not move (I'm doing this just by a cut/paste of the right part of the json file), but first use case is more tricky, because the references must be updated too to keep the same behaviors. eg. :
- Layer 1 has a `move to layer 2` key
- Layer 2 is my numpad layer
- If I move Layer 2 to Layer 4, I want the `move to layer 2` of my layer 1 to be updated to a `move to layer 4`, and keep move to my numpad layer.

Clearly, the drag & drop dance will be awesome :-)

algernon, 2022-06-03:
I don't think we should be updating the keymap to automatically update layer keys. There are too many edge cases to consider. So what I plan to implement is a simple reorder, without touching the keymap at all.

If we had layer _names_, then we could keep the keymap in sync with the moving around. Until we have that, I'd rather not touch the keymap when moving layers, because layer numbers are meaningless without knowing the intent behind the layer keys. Having a name associated with a layer would make the intent clear, and would allow us to auto-update keymaps. But that's a different feature, not in scope for the first go at layer reordering.

Rom1deTroyes, 2022-06-03:
Yes something more semantic would be awesome, but will eat some bytes of memory !

Swapping things, with no border effects will, indeed, be safer :+1:

algernon, 2022-06-19:
I had a go at this the other day, and it is a bit tougher than first anticipated. Primarily because the keymap and the colormap are separate entities in Chrysalis at the moment, yet, when reordering layers, we need to reorder both. In practice, this means that before we can reorder layers, we need to change how Chrysalis handles the keymap and the colormap.

Technically, we could do it without that, but I feel it would be too fragile. So this is going to have to wait until we refactor Chrysalis internally a bit. But once we're done with that, making layers drag-and-drop rearrangable will be a breeze.