keyboardio/Chrysalis #582: "No polkit authentication agent found" on Manjaro Linux

murchu27, 2020-10-10:
**Describe the bug**
Chrysalis does not seem to be able to find my polkit agent, and does not give me any option to point it to it.

**To Reproduce**
If the user running Chrysalis doesn't have access to the device file for Atreus (e.g. `/dev/ttyACM0`), Chrysalis will offer to add a `udev` rule (see screenshot 1 linked). Presumably sudo permissions are needed for this change, as Chrysalis tries to launch a polkit agent. However, it can't seem to find the one on my system (screenshot 2).

https://imgur.com/a/Z9Kbzgm

**Expected behavior**
Chrysalis should add the `udev` rule as intended by the function.

**Desktop:**
 - OS: Manjaro Linux
 - Chrysalis Version: 0.7.9+268

algernon, 2020-10-11:
@murchu27 Can you try if running `pkexec /bin/bash -c 'echo foo'` works, or if it produces a similar error?

For context, we're using [sudo-prompt](https://github.com/jorangreef/sudo-prompt) for elevating our privileges, which in turn will try to run pkexec. It seems it's... having issues doing so, and I'd like to figure out which one of the existing sudo-prompt issues is biting us here.

murchu27, 2020-10-11:
Thanks for the suggestion @algernon. I tried running the above, and after putting in my password, I got this line:
```polkit-agent-helper-1: error response to PolicyKit daemon: GDBus.Error:org.freedesktop.PolicyKit1.Error.Failed: No session for cookie
==== AUTHENTICATION FAILED ====
Error executing command as another user: Not authorized

This incident has been reported.```
Maybe there's something wrong with my setup?

algernon, 2020-10-11:
Mhm, it seems there's no polkit agent running indeed. Are you using a desktop environment, or something more custom? Most DEs will have a polkit agent, which should be running for pkexec to work.

murchu27, 2020-10-11:
I'm trying out awesome-wm at the moment but I also have i3 installed. I logged in to i3, and sure enough, the command worked! I'll try logging in with another user who doesn't have permissions, and see what happens...

murchu27, 2020-10-11:
Okay, getting another error now. I opened Chrysalis as a user that wouldn't have permissions on the device file, and the error came up saying they didn't have permissions, and offered to write the `udev` rule. When I clicked "Fix It", I got a password prompt, so that part's all good.

Then, I got this error:
```
Error: Command failed: cd "/home/fi_cradz"; "/usr/bin/pkexec" --disable-internal-agent /bin/bash -c "echo SUDOPROMPT; (mv /etc/udev/rules.d/60-kaleidoscope.rules /etc/udev/rules.d/60-kaleidoscope.rules.orig || true) && cp /tmp/.mount_Chrysa2b5Q2t/resources/static/udev/60-kaleidoscope.rules /etc/udev/rules.d/ && udevadm control --reload-rules && udevadm trigger /dev/ttyACM0" mv: cannot stat '/etc/udev/rules.d/60-kaleidoscope.rules': No such file or directory cp: cannot stat '/tmp/.mount_Chrysa2b5Q2t/resources/static/udev/60-kaleidoscope.rules': Permission denied
```
Looks like the template file that Chrysalis wants to copy to `/etc/udev/rules.d` is having issues. Check out this `ls` entry:
```
d?????????  ? ?        ?             ?            ? .mount_ChrysahdHDLB
```
(the full name of the folder is a little bit different than the name of the folder in the error, but it looks like those are just random characters generated when Chrysalis launches. I had exited Chrysalis by the time I thought to check for that `ls` entry, hence why it has different characters.)

algernon, 2020-10-11:
Good, good, making progress! I think I can fix this one from here, will push a possible fix later today.

cytommi, 2020-11-24:
I'm also facing the same issue.... has this been fixed? Thanks!

algernon, 2021-01-25:
It appears that #603 has been a duplicate of this one. I just fixed the installation problem on master. The 0.8.1-snapshot build should have the fix, as will the upcoming 0.8.1 release.

closed 2021-01-25 by algernon

pheuberger, 2023-02-20:
I have the same error (sway-wm on fedora 36). can i just manually set permissions before starting Chrysalis?