keyboardio/Chrysalis #583: Generate printable or ecporp

James-Firth, 2020-10-11:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

gspringtech, 2020-10-19:
I would also like a printable key layout. This _could_ be achieved using the SVG generated by the Keymap.js. I have tested by logging ReactDOMServer.renderToString(layer) in the Editor.js - but I'm not a React expert so not sure how to progress from there. The SVG would need a stylesheet added to change the text and fill colours.

obra, 2022-05-21:
This is now available in the latest snapshot release.

closed 2022-05-21 by obra