keyboardio/Chrysalis #585: Can't launch from macos

joelp1, 2020-10-11:
**Describe the bug**
I cannot start the program in macos 10.15.6 (19G2536)

I get the following error:
“Chrysalis” cannot be opened because the developer cannot be verified.

Steps to reproduce the behavior:
1. Download https://github.com/keyboardio/Chrysalis/releases/download/chrysalis-0.7.9/Chrysalis-0.7.9.dmg
2. extract
3. move to apps
4. eject image
5. launch chrysalis from Launchpad

**Expected behavior**
chrysalis starts.

**Screenshots**
If applicable, add screenshots to help explain your problem. If possible, please also include the debug bundle you can create on the "System Information" screen. If, for some reason, you are unable to access that screen, please press `Control + Shift + I` to bring up the developer console. Right-clicking inside it will allow you to save its contents to a file - please attach that to the bug report.

**Desktop (please complete the following information):**
 - OS: macos 10.15.6 (19G2536)
 - Chrysalis Version: 0.7.9

**Additional context**
Add any other context about the problem here.

obra, 2020-10-11:
Can you tell me if
http://kaleidoscope-builds.s3-website-us-west-2.amazonaws.com/Chrysalis/latest/Chrysalis.dmg
works any better for you?

joelp1, 2020-10-11:
> Can you tell me if http://kaleidoscope-builds.s3-website-us-west-2.amazonaws.com/Chrysalis/latest/Chrysalis.dmg works any better for you?
> […](#)

Yes, that artifact works for me.

joelp1, 2020-10-12:
I also tried installation via brew cask and ran into the same issue.

smsm1, 2020-10-14:
This is due to the macOS security and requirements to have the apps signed. You can still open the app when download e.g. via homebrew by right clicking the app and then choosing "open". The first time you'll get this dialog:

[image]

Click cancel and then do it again when you'll get this dialog instead:

[image]

The first step can be done by double clicking the app rather than using the right click menu to the same effect. This is the same for several apps installed via e.g. homebrew.

tilgovi, 2021-09-29:
It would still be lovely to fix this if it's not a total pain. I can't run the app at all on my corporate-controlled macbook.

obra, 2021-09-29:
Current builds *are* signed. I don't know if Homebrew is grabbing our
signed builds or building them themselves. But let me know if first-party
builds are still failing like this for you, ideally with the exact error.

algernon, 2021-09-29:
> Current builds *are* signed. 

Sadly, they're not, not since we switched from Travis to GitHub Actions:

https://github.com/keyboardio/Chrysalis/runs/3725908378?check_suite_focus=true#step:9:2087

That's an unfortunate regression we haven't restored yet.

obra, 2021-09-30:
This should now be fixed. Can you folks try
https://github.com/keyboardio/Chrysalis/releases/download/snapshot/Chrysalis-0.8.6-snapshot.72.dmg
and see if it works?

tilgovi, 2021-09-30:
That works! Thank you!

algernon, 2021-10-02:
This is resolved for snapshots, and will be resolved for stable builds too, as soon as I tag the next one (in a couple of minutes). Thanks for testing the snapshot build!

closed 2021-10-02 by algernon