keyboardio/Chrysalis #588: FR: Making simple key combo macros

cablebox, 2020-10-12:
There's a couple of 3-4 key combos I press regularly in my workflows. However, I don't have any programming experience, and the documentation on how to build macros is a bit thin at the moment. All I want is a GUI experience so I can say 'when I press this button, send these key codes'

Rom1deTroyes, 2020-10-12:
If it's a combo (eg. : <kbd>Ctrl</kbd>+<kbd>Alt</kbd>+<kbd>Shift</kbd>+<kbd>O</kbd>), you can do it with Chrysalis : just select the modifiers you want to enable with your key.

If you want a sequence of letters (eg. : <kbd>:</kbd><kbd>w</kbd><kbd> </kbd><kbd>/</kbd><kbd>t</kbd><kbd>m</kbd><kbd>p</kbd><kbd>/</kbd><kbd>t</kbd><kbd>m</kbd><kbd>p</kbd><kbd>.</kbd><kbd>m</kbd><kbd>d</kbd><kbd>Enter</kbd>, you need to write a macro in Kaleidoscope, compile and flash your keyboard, than assign your macro to a key in Chrysalis. That's not so hard to do by following the example given by the Macro whom write the information about the keyboard.

Happy Hacking !

runrin, 2020-10-30:
I second this feature request. It kind of seems like a missing feature in Chrysalis, as nearly every other keyboard software has the ability to make and control macros.

Perhaps better, a way to enter a macro without Kaleidoscope or Chrysalis, using a key combo to record, and set macros keyboard side.

jacereda, 2020-10-31:
I want to have a layer containing accented characters (composition of dead accent key + letter).  Something like this would definitely help.

stevep99, 2021-05-24:
Yes please!  Just want to add my voice to those requesting this feature. I also want easy dead-key/compose strings.

Having to "write a macro in Kaleidoscope, compile and flash your keyboard, than assign your macro to a key in Chrysalis" is not a very user-friendly feature.

algernon, 2021-05-25:
For what its worth, the Kaleidoscope-side of this feature is complete (but hasn't seen much testing), so all we need is to build an UI on top of it, and include the plugin in the firmware shipped with Chrysalis.

It's something reasonably high on our priority list, and I have a couple of ideas how to make a nice UI for it, but didn't get to the point yet where I can share mockups.

AshitakaLax, 2021-08-05:
Sweet stuff @algernon I look forward to testing it out.

algernon, 2022-05-27:
We're now very close to having editable macros within Chrysalis, #703, the pull request adding it, is in a good state, and offers enough functionality to create simpler key combo or sequence macros.

It doesn't cover every case yet (compose + accents for example, those will be problematic if the accent needs shift, and you want to type an upper-case accented character), but comes pretty close. We'll continue iterating over the feature, to cover use cases the initial implementation may not cover yet.

algernon, 2022-05-28:
The macro editor is now on master, and should be in the next release (and the next snapshot too, before that).

closed 2022-05-28 by algernon