keyboardio/Chrysalis #590: Add BÉPO in "Import/Export the current layer" defaults

xenlo, 2020-10-13:
## Description
It would be great (and quite easy I think) to add a BÉPO layer as possible default load in "Import/Export the current layer". It will help a lot the BÉPO to start with the keyboard.

## Solution
I had a look a bit already, so I think that add `bépo.json` in `static/Keyboardio/Atreus` and a link to it (in `src/renderer/screens/Editor/ImportExportDialog/LoadDefaultKeymap.js`?) should do the work.

## Alternatives
There was already some kind of initative to share a BÉPO layer on http://bepo.fr/wiki/Atreus.

## Additional info
The problem I have to start a PR answering myself this Feature Request is that I don't know if I have to start with the assumption that people will have their system local keyboard map set to `us` (qwerty) or `oss,bépo` (BÉPO). which change completely the map between keycodes and labels.

xenlo, 2020-10-13:
After a chat with @algernon, a major code improvement is foreseen which will ease the set-up of beginner user who has other system keymap that US-QWZERTY!

lobre, 2020-10-14:
Apart from point of view of the Chrysalis software, I’d be interested in trying to find what would be a sane default bépo configuration for the Atreus. After having thought a while about it, I feel any solution would be quite opinionated as there are a lot of keys to remap with this 44 keys limitation.

RomainChamb, 2020-10-22:
I would be quit interested in having a default BEPO configuration for the Atreus. I try to create one but fails as of now.

xenlo, 2020-10-22:
I spend few hours last night trying to forge a [bépo layouts for the Atreus](https://github.com/xenlo/atreus_layouts), and that was not easy. My through about that is that 44keys is a bit too short for bépo.

Nevertheless, back to Chrysalis, it's actually pain in the choobies, that each time we import a layout from a json where I updated the labels (to match bépo keys), the preview is right but one save it, it flush all your nice labels! :cry:

lobre, 2020-10-22:
@obra, it's seems to be quite a challenge to get there, but it's awesome that there is current work to reach that goal! I guess it will be way easier to play around with layouts such as bepo when this will land.

Effectively, it is harder to play with the keyboard without this feature for us (non us people). 

Thanks @xenlo for sharing your configuration. I did not know that bepo.fr had a page for the atreus. That's good news. I am sure we will have a common bepo layout on which people will rely in the future.

RomainChamb, 2020-10-26:
@obra Really cool to read that you are working on it ! Thanks for you amazing work

lobre, 2021-01-26:
Now that 0.8.0 is out, I guess it will be easier to configure a layout for bépo and to share it. Maybe that's time to revamp what has been shared on http://bepo.fr/wiki/Atreus and propose a full export.

I personally like the proposition [here](https://bepo.fr/wiki/Fichier:Atreus_layer_0_2020-10-23.png) even if I would switch a few modifiers.
- bksp at shift location (I use it more often than del, so I prefer having it on the bottom row)
- del at bksp location
- shift at del location (I like to have modifiers on the sides, and bksp/space second keys from the sides)
- AltGr at Super location (AltGr is on the right on a traditional azerty keyboard)
- Alt at AltGr location (I feel easier to type Alt + Tab when the two keys are not right beside each other)
- Super at Alt location (Super is on the left side of a traditional azerty keyboard, just at the left of Alt)

I know we will not find a layout that satisfies everyone, but having a "default" one to share might help as a starting point for people that want to tweak it a little bit.

lobre, 2021-04-04:
I have spent time this week to imagine a Bépo configuration that would make sense, trying to avoid adding too many opinions to it.

I have documented the process and result on this GitHub repo: https://github.com/lobre/atreus-bepo

My hope is that it can serve as a starting point for someone that wants to use Bépo on the Atreus (as it is not super usable out of the box).

The repo contains the `.json` exported layout.

I’d like to receive feedback on that. I am thinking that if we can reach a consensus at some point for a basic Bépo layout, maybe it will make sense to propose it as a default one alongside "Colemak", "Dvorak" and "Qwerty" in the Layout sharing screen.