keyboardio/Chrysalis #597: Import and export all layers in a single operation

thoni56, 2020-10-24:
**Is your feature request related to a problem? Please describe.**
I often make changes to multiple layers. I really want to make sure that I can get back to a configuration later, but unless I am missing something I now have to export each layer separately. This is cumbersome.

**Describe the solution you'd like**
I'd like to be able to save a complete key configuration including all layers and be able to import that as a single operation.

**Describe alternatives you've considered**
As I can only export one layer at a time I considered, and probably will have to, storing all layers in a folder and name the folders after what I'm changing. No problem if this is what the solution does, In that case I'd like to only need to select the folder.

**Additional context**
As I'm Swedish I want to customize the layout to suit my language (there are other features I'll add some issues for).

jpjerkins, 2020-12-29:
Was thinking I was exporting my entire configuration, and was checking the changes into a local git repo.  Finally looked at the diffs, and realized each commit was saving a different layer.

This enhancement would both make it easier to save/restore an entire layout, but would also make it trivial for the community to share keymaps and keymap ideas, learn from each other, and drive innovation beyond the tired QWERTY/Dvorak/Colemak debate.  Let's enable a discussion around the _entire keyboard!_

obra, 2020-12-29:
If you you grab the latest dev build, you should already see this live.

jpjerkins, 2020-12-29:
Awesome! Thanks for the tip! And I apologize for the spam!

Phil Jerkins

algernon, 2021-01-25:
As this is available both on master, and in the latest release, I'm closing this.

closed 2021-01-25 by algernon