keyboardio/Chrysalis #598: Support and display international keys when using a non-US-English setting in the OS

thoni56, 2020-10-24:
**Is your feature request related to a problem? Please describe.**
As I'm Swedish I want to customize the layout to suit my language. I of course have my OS keyboard setting set to Swedish. This means that when I want to move a key in Chrysalis I first have to (manually, mentally) map that key to what it displays on the American keyboard and then find that key in the Chrysalis UI. Later when I remind myself on what key to use I need to do the reverse mental operation. Very unfriendly to non-US people...

**Describe the solution you'd like**
Make it possible to display and configure keys using its localized setting.

**Describe alternatives you've considered**
Not sure there are any, except the manual double-two-way translation.

**Additional context**
Related to this is of course that many international characters are actually found under "Separators" since e.g `[` and `{` are the representation of `Å` and `å` on a Swedish keyboard.

eldarintar, 2020-10-31:
I am french and I have a similar issue. Especially I don’t want an AZERTY layout, but rather a BÉPO layout (equivalent to the DVORAK for french), which adds to the complexity.

eloyesp, 2020-12-06:
If translation maps can be defined, it would be awesome to make it easy to apply them on the keyboard, may be reversed, to nullify changes made by the OS, with something like a composition layer.

lobre, 2020-12-10:
Duplicate of https://github.com/keyboardio/Chrysalis/issues/513

thoni56, 2020-12-10:
That issue had toooo looong a title to be visible when I looked through them...

closed 2020-12-10 by thoni56