keyboardio/Chrysalis #599: the board seems to get into an inconsistent state when moving between layers

rca, 2020-10-27:
**Describe the bug**
I have a key combination where the `layer shift` key on the base layer shifts to `layer 1` and `layer 1` contains a key to lock to `layer 2`.  `layer 2` has a key to lock to `layer 0`.  When moving back to `layer 0` some of the keys fail to remap.  For example, when this happened to me just now, the `y` key would not type the letter but most of the other keys were remapped properly where I could mostly write, but at least the `y` key wasn't typing the letter.

**To Reproduce**
Steps to reproduce the behavior:
1. from `layer 0` shift to `layer 1`
2. on `layer 1` hit the key to lock to `layer 2` (my layer 2 contains arrow keys, plus page up, page down, home, and end mapped on the right half of the keyboard.
3. perform some navigation keystrokes on `layer 2`
4. on `layer 2` hit the key to lock to `layer 0`
5. on `layer 0` type some words; most keys will be back to letters, but some keys will not remap properly

**Expected behavior**
Hitting the button to lock back to `layer 0` properly remaps the keys to reflect the selected layer

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS: MacOS 10.15.7
 - Chrysalis Version: 0.7.9

**Additional context**
When the keyboard gets into this in-between state, I have resorted to simply disconnecting the keyboard and reconnecting it; which will reset it to a known state.

I have also been able to repeat the steps to switch back to `layer 2` and hit the key to lock to `layer 0`.  In my keyboard's current state I have been able to reproduce 3 times that the remapping does not completely happen when locking back to layer 0.

rca, 2020-10-27:
I just noticed there are two options that appear very similar, "moving to" a layer and "locking to" a layer.  I have switched my mapping so that I use the "move to" option to switch between layers and that appears to fix my particular use-case.

Is there documentation on the difference between these two options?

Thank you.

algernon, 2020-10-27:
The documentation about the various layer keys is [here](https://kaleidoscope.readthedocs.io/en/latest/layers.html#layer-keys).

As for the issue, it's not really a bug, but simply how layer stacking works. With `Lock to layer` and `Shift To Layer`, any layers that were previously active, remain active. So when you go 0->1->2->0, the layer stack will be 2->1->0 (or 2->0, if you released the shift-to-layer key meanwhile), because `LockLayer(0)` does _not_ deactivate layer 2. MoveTo does, which is why that fixes your use-case.

In general, if you don't care about layer transparency, if you don't use the properties layer stacking would give you, then move to layer is usually the better choice, instead of LockLayer.

rca, 2020-10-28:
I see and now understand the difference between locking to and moving to; thanks!

I still don't understand why the key mapping would be in an inconsistent state, however.  As I understand it, locking to a layer where all the keys are defined should use the definition of the last activated layer.  In my case locking to layer 0, where all the keys are defined, should take priority over any previous layer, right?

algernon, 2020-10-28:
Without knowing your full keymap, there are two possibilities for the strange state, as far as I see. Before I dive deeper, a few things of note: first, a layer cannot be activated twice, so going 0->1->2->0 will not have 0 on top, because it was already active, so it remains on the bottom. Second, due to technical and historical reasons, locking and unlocking a layer is done by the same keycode, it's more of a toggle, really. So if you have layer 0 active, and you press a `LockLayer(0)` key, that will deactivate layer 0. Mind you, the firmware will still fall back to layer 0 if it can't find a key elsewhere, so it's still semi-active, but it will not be on top. In either case, we'll end up in a situation where the lookup order is 2->1->0, rather than 0->2->1. By using Move, the other layers get deactivated, so there's only layer 0 to look up from.

rca, 2020-10-28:
> first, a layer cannot be activated twice, so going 0->1->2->0 will not have 0 on top, because it was already active, so it remains on the bottom.

This sounds like the issue I am running into.  Thank you for the detailed response!  For my current use-case, it sounds like I want to use move instead of lock.

Thanks again!

closed 2020-10-28 by rca