keyboardio/Chrysalis #601: Layer shift when held not working

90bookwyrm, 2020-11-03:
**Describe the bug**
The layer shift when held, normal key otherwise toggel does not function.  

**To Reproduce**
Steps to reproduce the behavior:
1. map the function button to enter with the layer shift toggle selected to layer 2
2. save changes
3. press and hold the newly mapped button and try to use layer two buttons
4. See error : no layer two function

**Expected behavior**
pressing and holding the button mapped to layer shift when held should result in a temporary layer shift and the ability to use layer 2 functions. ie pressing j should yield moving down a line not the letter j

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: iOS 10.13.6
 - Chrysalis Version: 0.7.9

**Additional context**

Rom1deTroyes, 2020-11-03:
I've just tried on the `master` branch with my GNU/Linux laptop and it works as expected...

Does your left palm key got the second layer working ?
And just to be sure, you are trying to <kbd>Enter</kbd>+<kbd>j</kbd> with your right palm key, not one of the other <kbd>Enter</kbd> key ? (things can be confusing sometimes :-)

90bookwyrm, 2020-11-09:
> And just to be sure, you are trying to Enter+j with your right palm key, not one of the other Enter key ? (things can be confusing sometimes :-)

Hi Roman,

I am working on a Mac laptop.  The left function key works just fine as does mapping any key to shift layers.  It is only the shift to layer toggle that does not work. I I am using the key that I mapped to enter with the toggle layer shift. I have tried it with different keys and mappings, but the toggle never works.  Looking at the community page, I am not the only one to have run into this...

https://community.keyboard.io/t/layer-shift-when-held-not-working/3565

Rom1deTroyes, 2020-11-09:
The solution may be there :

>    algernon Gergely Nagy Feb 20
> It’s possible to do it via Chrysalis, but you need to use the experimental (with bells & whistles) firmware, instead of the default one.

Have you tried to update the firmware ?
(you can do this with Chrysalis, and come back to the stable firmware by the same way)

algernon, 2021-01-25:
Layer shift when held does work when using the experimental firmware. We do need better explanations about what feature requires special firmware and which features do not, but that's a separate issue, I believe.

closed 2021-01-25 by algernon