keyboardio/Chrysalis #603: Linux Keyboard Permission Issue "Fix It" Button Doesn't Work

rjhilgefort, 2020-11-11:
**Describe the bug**
On the [development build](http://kaleidoscope-builds.s3-website-us-west-2.amazonaws.com/Chrysalis/latest/Chrysalis.amd64.AppImage), the "Fix It" button gave me an error that looks like permission issues.

I was chatting with @obra on Discord and that conversation can be seen here:
https://discord.com/channels/492408953041321984/638783537734090813/775867293196484609

**To Reproduce**
Steps to reproduce the behavior:
1. Download development build
2. Click on "Fix It" related to udev rules

**Expected behavior**
I suppose I expected it to resolve the permission issue it claimed to say it would fix.

**Screenshots**
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Arch 5.9.6
 - Chrysalis Version: `0.7.9+286`

**Additional context**
N/A

algernon, 2020-11-11:
I guess Arch is not using udev and/or systemd?

obra, 2020-11-11:
can we detect the failure we ran into and point people at a webpage? possibly showing the user which device you need access to, what it’s permissions are and What group you would need to be in to be able to access it. On the theory that if the smart way to do it has failed Users probably need to resort to getting into the correct group

algernon, 2020-11-11:
We can, yes. We can't (easily) detect the exact error, but we can direct them to a page when we encounter any error. Apart from authentication, we can detect that, and we shouldn't direct them anywhere in that case.

kurzkopfgleitbeutler, 2021-01-17:
I get the same issue with Chrysalis-0.8.0.amd64.AppImage on Ubuntu 20.04.1 LTS.

```Error: Command failed:
cd "/home/foo/chrysalis";
"/usr/bin/pkexec" --disable-internal-agent /bin/bash -c "echo SUDOPROMPT;
(mv /etc/udev/rules.d/60-kaleidoscope.rules /etc/udev/rules.d/60-kaleidoscope.rules.orig || true)
&& cp /tmp/.mount_ChrysaxM57yH/resources/static/udev/60-kaleidoscope.rules /etc/udev/rules.d/
&& udevadm control --reload-rules
&& udevadm trigger /dev/ttyACM0"

mv: cannot stat '/etc/udev/rules.d/60-kaleidoscope.rules': No such file or directory
cp: cannot stat '/tmp/.mount_ChrysaxM57yH/resources/static/udev/60-kaleidoscope.rules': Permission denied
```

From the error, I'd guess it has to do with how permissions in AppImage are handled. Maybe (n.b. I don't know about this kind of stuff) it has to do with chromium sandboxing or preserving ownership in the AppImage, does this have to do with it?
https://github.com/laurent22/joplin/issues/2246#issuecomment-577315584

hatxet, 2021-01-24:
@kurzkopfgleitbeutler I get the same error as you, also on Ubuntu 20.04 LTS. That temporary mount folder in `/tmp` sure doesn't look right.

```
root@localhost# ls -la /tmp/                                                                                                                                                   
d?????????  ? ?     ?          ?            ?  .mount_ChrysatuM5zq/  

root@localhost# ls -l /tmp/.mount_ChrysatuM5zq                                                                                                                                                                      
ls: cannot access '/tmp/.mount_ChrysatuM5zq': Permission denied
```

algernon, 2021-01-25:
Reproduced, fix tested & pushed to master.

commit 4a2c4be4 (2021-01-25): Fix the installation of the udev rules

When installing the udev rules from the AppImage, we need to copy the rules to a
temporary place first, before elevating privileges, and then copy from there to
the final place. We need to do this because the AppImage temporary mount is
locked down, and not even root can access it, so we need to get the source file
out of there first.

Fixes #603.
closed 2021-01-25 by algernon