keyboardio/Chrysalis #604: Sharing layouts and votes

artkpv, 2020-11-12:
I'd like to see how other setup their keyboards so that we can find a better configuration or find new ways how to press keys.

My current issue. Received Keyboardio Atreus about two weeks ago. I'm now learning to use it. Not a simple process. Frequently I stop to think about how to press such combinations with f1, f2, esp. alt+f3, alt+f4, or just those f0 keys. I modified my keyboard to swith to layer 2 where those keys are using fun+esc so that when i press fun+esc+3 or 4 and alt, 4 keys to press, I get what I want. I wonder if there a better way, an easier way to setup such combinations. Or combinations with shift key.

Solution might be some some chart with votes which layout the best. Some way to publish my keyboard so others can see.

jpjerkins, 2020-12-29:
Would also love to have this feature.  Would be a great help in convincing other programmers that something as minimalist as the Atreus is not just workable, but actually _helpful_ for those who write code.

obra, 2020-12-29:
Yes. Once we get multilingualizaiton landed, this will be one of the highest priorities for Chrysalis.

jpjerkins, 2020-12-29:
Awesome. It’s awesome that you already have this on your roadmap, and it’s great just to hear that you have a roadmap :-)

Phil Jerkins

obra, 2020-12-29:
"Roadmap" may be a strong phrasing. We want to do layout sharing next.
After that, probably macros in Chrysalis.