keyboardio/Chrysalis #605: OneShot layer canceled by Modifier-on-Hold

billba, 2020-11-12:
**Describe the bug**
On my Atreus, after tapping a OSL key once, using a modifier-on-hold key cancels the layer

**To Reproduce**
0. Use an Atreus
1. set a key as OSL - In my case [fun] is OSL for layer 1
2. set another key for modifier-on-hold - in my case [Z] is shift on hold
3. have a target key in the designated layer that needs said modifier - in my case the [Q] key is bound to [1] and I'm trying to get [!] via shift
4. Tap the OSL key [fun] then hold down the modifier-on-hold key [Z] and press the target key [Q]

**Expected behavior**
Expect "!" but instead get "Q" -- in other words, the initial switch to layer 1 was canceled

**Desktop (please complete the following information):**
 - OS: Mac 10.15.7
 - Chrysalis Version: 0.79+286

**Additional context**
Works if I *hold down* the OSL key, e.g. [fun]+[Z]+[Q] => "!"

gedankenexperimenter, 2021-06-02:
I haven't confirmed that I can reproduce this with the old version of Kaleidoscope from Chrysalis 0.7.9, but it works properly in my tests of the current master version of Kaleidoscope.

algernon, 2022-05-17:
This should be fixed by updated firmware, so closing it.

closed 2022-05-17 by algernon