keyboardio/Chrysalis #609: Chrysalis + Atreus + Move to layer exposes nonexistent layers 10-31

parke, 2020-11-21:
When configuring a Keyboardio Atreus with Chrysalis, I am given options to:

"Shift to layer"s 0-9, and
"Lock layer to" 0-9.

But I can "Move to layer"s 0-31.  (Even though the Atreus apparently only has layers 0-9.)

If you need a more detailed bug description/instructions, please let me know.

Rom1deTroyes, 2021-01-22:
Can't reproduce this with my k.io Atreus and Chrysalis 0.8.1-snapshot under Debian GNU/Linux.

Way tested :

1. Click on a blank key
2. Under `Layer and keymaps`, choose `Move to Layer`
3. Increase the numbers : you can't go higher than 9.
4. Write `10` on the number box, it is changed by a 9.

But the `Layer shift for next Action` seems limited to 7 (see #642).

parke, 2021-01-22:
It appears I probably encountered the problem with Chrysalis-0.7.9.AppImage.

Maybe the problem has been fixed since then.

algernon, 2021-01-25:
Indeed, this is something that's been fixed between 0.7.9 and 0.8.0. Closing this issues accordingly.

closed 2021-01-25 by algernon