keyboardio/Chrysalis #611: Number keys show their shifted symbols (e.g. `1` and `!`), while symbol keys do not (e.g. `=`, `[`)

rca, 2020-11-27:
**Is your feature request related to a problem? Please describe.**

When editing a layout in Chrysalis, the number keys show the symbol that they would print when the shift key is held down, while symbol keys do not.

**Describe the solution you'd like**

It would be nice to see the shifted key values for symbols just like they are visible for numbers.

algernon, 2021-01-25:
I believe this is now fixed on master, with the new editor & keycode database. Can you verify if the current snapshot  does what you think it should do?

aspiers, 2021-01-26:
It looks to me with 0.8.1 that the opposite is now true: *neither* the number keys nor symbol keys show what they will output when shift is held.

algernon, 2021-01-26:
Oooh, hmm. You mean that the _selector_ should show _both_? (As opposed to the keymap showing the shifted symbol if the key is mapped to shift+key)

aspiers, 2021-01-26:
I'm not sure what you mean by selector, but as per the original report this issue is about what is displayed on the graphical representation of the keycaps.

algernon, 2021-01-26:
By selector, I mean the 104-key keyboard that pops up when you press "Pick a key".

Nevertheless, the other part of my question is still relevant: should the keycaps display both `1` and `!` for a key mapped to `1`, (but only `!` when mapped to `Shift+1`)? Right now, they display `1` when mapped to `1`, and `!` when mapped to `Shift+1`. The reason behind this is that right now, we display what keys do on their own, without modifiers or anything fancy. If we added modifiers, we'd run out of space quickly, because we'd need Shift and AltGr displayed for some keys too.

aspiers, 2021-01-26:
[**@algernon**](https://github.com/algernon) commented on [January 26, 2021 6:09 PM](https://github.com/keyboardio/Chrysalis/issues/611#issuecomment-767727480 "2021-01-26T18:09:42Z - Replied by Github Reply Comments"):
> By selector, I mean the 104-key keyboard that pops up when you press "Pick a key".

OK, so we're talking about the same thing.

> Nevertheless, the other part of my question is still relevant: should the keycaps display both `1` and `!` for a key mapped to `1`, (but only `!` when mapped to `Shift+1`)?

Yes.  As witnessed by the title of this issue, this is how it was before your jumbo PR (which improved a ton of things BTW - congrats on that!)  And this makes sense because a) that's what you typically see on the keycaps so everyone understands what it means, and b) there's enough space to show it, and it makes life easier, so why not?

> Right now, they display `1` when mapped to `1`, and `!` when mapped to `Shift+1`. The reason behind this is that right now, we display what keys do on their own, without modifiers or anything fancy. [...]

Please let's not throw the baby out with the bathwater ;-)  Displaying the shifted symbols is better than not displaying them.  If there isn't room to display other stuff too, that doesn't invalidate the benefits of displaying the shifted symbols.

algernon, 2021-01-26:
Showing them on the 104-key selector is certainly something we can do, I have nothing against that. Can likely fit AltGr there too, for good measures.

It's the keyboard image where I'd be _much_ more conservative about what we display and what we don't.

aspiers, 2021-01-26:
OK so maybe I misunderstood.  I don't know what the difference is between the 104-key selector and the keyboard image.  All I know is that, as per this issue title, prior to the big redesign, the number keys showed the shifted symbols but the symbol keys didn't, and now neither do.  IMHO (and presumably in the eyes of the OP) this is a move in the opposite direction from what would be ideal.

Rom1deTroyes, 2021-01-26:
I have the same feelings here (from both of you :thinking: :
- if the 104 key picker is meant to be an image of a real keyboard, I expect to see the same symbols as my laptop keyboard (ie. <kbd>AZERTY</kbd>... for the letters, <kbd>&1</kbd><kbd>é2~</kbd><kbd>"3#</kbd>... for the numbers, <kbd>)°]</kbd><kbd>=+}</kbd><kbd>^¨</kbd><kbd>$£ê</kbd>... for the symbols, etc.) : the cargo picker is an awesome improvement here,and choosing the system layout is a life changer !
- On the main keyboard image (the one with the Model01 or the Atreus), I expect to see just what will happen when I press the key, so per layer representation as it is now seems very intuitive for me, but needed to be tweaked at the corners.

For now, if I want to have a layer with the legacy numbers row splited like numbers on the first row and symbols on the second row, and my system is in AZERTY, here is the screen I have :

Chrysalis-0.7.7 :
- the letters on the Atreus keyboard are from the QWERTY keyboard,
- the numbers (shifted in AZERTY) show both lower and upper character.
[image]

Chrysalis-0.8.2-snapshot :
- the numbers (shifted in AZERTY) on the Atreus keyboard are the shifted symbols from QUERTY, and on the 104 picker the upper symbols are not shown (so you can't find `+%?./`.
- there is also problem with <kbd>éèçà</kbd> that are shown in an upper font, but are not shifted (`É` is <kbd>AltGr</kbd>+<kbd>Shift</kbd>+<kbd>é</kbd> on my GNU/Linux box).
[image]

I don't know if you prefer having all the ideas and feedbacks here to think about it, or splitting things on others issues to clarify all this assaults :-)

Amazing work, BTW !

obra, 2021-01-26:
I agree that we should show (at least) shifted symbols (but not both upper and lower case letters) on the key picker.

Rom1deTroyes, 2021-01-26:
Yes, showing both <kbd>aA</kbd>, <kbd>bB</kbd> <kbd>cC</kbd> is very heavy and don't add simplicity !

Some examples, for reference ; 

- A heavy but complete technical picture : [image]

- a complete keyboard, without lower case when no need to : [image]

- a good compromise between light and complete : [image]

- Some users may need a more than complete option : [image]

- An other way to look at layers on a standard keyboard : [image]

aspiers, 2021-01-27:
Ah OK, now I finally understand what you mean by the selector / picker.  I hadn't noticed that until these comments.  But this issue was originally about the main keyboard, before this picker even existed.  So while it's great to discuss both, please let's not lose sight of the original issue and the fact that the recent redesign has moved in the opposite direction from what the original report was asking for.