keyboardio/Chrysalis #612: Import/Export button tooltip and description

esden, 2020-11-30:
As the developer build now exports/imports all layers (🎉 ) in the `Import/Export` page the tooltip and the page description is not up to date any more.

It should be updated from "Import/Export the current layer" to something like "Import/Export keyboard layout" or something to that effect.

algernon, 2021-01-25:
We have recently reworked the whole import/export thing, and it no longer says anything about layers. It... doesn't really tell anything else either, so we still need some work on that front...

Moilleadoir, 2022-10-04:
The button label is currently `Import` and the couple of times I’ve had to load from a file or backup it’s taken my brain a while to figure out that there’s a button I need to press.

The problem is that there are three buttons in a column, so `Import` looks like another file action.  The headings of the other sections include the word `Load` (`Load from library`, `Load from backups`) so it would make sense for me to have the ‘action’ button separated a bit and labelled `Load` or `Load selected` (even though maybe Import is more accurate).