keyboardio/Chrysalis #618: Exports of system information shouldn't include localized strings

obra, 2020-12-08:
@algernon asked me to open this bug report.

algernon, 2021-01-25:
We no longer include localized strings. We do include a number of useless information, such as the keymap SVG in some cases. I'll keep this open, but we're looking at filtering out the useless stuff from there, rather than localized strings in particular.

algernon, 2021-01-25:
Hmm.... it looks like we only include those SVGs in the `found_devices` log messages, that was easy to filter out. Not sure if that's the best way, but it was easy, and only a few lines. A proper solution (completely separating the device info object and the SVG component) would've taken much more effort for very, very little gain.

commit 4599d2fa (2021-01-25): api/focus: Filter out the keyboard SVGs when logging found devices

Fixes #618.
closed 2021-01-25 by algernon