keyboardio/Chrysalis #620: Locking / Moving to active Layer does not work

tnovotny, 2020-12-20:
**Describe the bug**

On layer-0 I modified [space] to be [L#3/] (layer shift when held, normal key otherwise)

[image]

On layer-3 i have created a numpad on the right side

[image]

I want to modify the [Y] button to keep the layer active after I release the [space] key. When I set it to either [move to layer 3] or [lock to layer 3]

[image]

After the release of all buttons, I am back in level 0.

However, when I change the level to a value that is different from 3, say 4, then it works as expectd

**To Reproduce**

1. Modifiy th layout as described above
2. press [space]-[Y]
3. press one of the numpad keys
4. the corresponding lower case letters appear

**Expected behavior**

I expected the layer 3 to still be active and the key press to result in a digit.

 - OS: Win10 20H2
 - Chrysalis Version: 0.7.9

**Additional context**

As a workaround, if I copy level 3 to level 6, and change [Y] to [lock to layer 6] or [move to layer 6] then it works as expected. It appreas that the release of [space] unconditionlly removes the layer shifted to whereas other layers remain untouched. 

Might relate to [OneShot layer canceled by Modifier-on-Hold #605](https://github.com/keyboardio/Chrysalis/issues/605)

algernon, 2021-01-25:
This is unrelated to #605, as far as I see. The problem is on the Kaleidoscope-side, nevertheless: we currently only store the active layer stack, we do not "remember" how the layers were activated. So when you hold `[L#3 / Space]`, we activate layer 3. When you press `[Move to Layer #3]`, we do nothing, because layer 3 is already active. Then, when you release `[L#3 / Space]`, it deactivates L3, because we do not remember that `[Move to Layer #3]` was also pressed.

What we might be able to do is that when we explicitly move to a layer, we make sure that any layer shifting keys currently pressed will be no-op when released. Mind you, this would only affect `Move To Layer`, because that disables any other active layers. I do not think it would be safe to do the same for `Lock To Layer`.

I'll open an issue on Kaleidoscope about this, and link it here.

algernon, 2021-01-25:
I'll keep this issue open until we know what happens with keyboardio/Kaleidoscope#1009, because if implemented, it will be fixed in a Chrysalis context when we update the shipped firmware files to ones with the change in place.

necsoiu, 2021-02-03:
I think we need to consider blocking the layer rather than moving on to it.
In other words when we are shifted to a layer we must to have a way to block that layer pressing one key (e.g. lock to layer key). Pressing again the same key must unlock the shifted layer.
I think this is an absolutely necessary feature on a keyboard with so few keys, otherwise we will have to use(block) a key on level 0 which we will use to lock and unlock a certain level.
In fact I think the title of this bug should be '**Locking/unlocking the active shifted layer**'.

guishogt, 2021-04-22:
Was this resolved? Having the same issue. 

Thanks.

gedankenexperimenter, 2021-06-02:
There is a [PR open against Kaleidoscope](https://github.com/keyboardio/Kaleidoscope/pull/1039) that would address this issue, if accepted.