keyboardio/Chrysalis #621: Add ability to print layout

jpjerkins, 2020-12-29:
**Problem Statement**
There currently isn't any way to print your custom layout with Chrysalis.  While I certainly appreciated the laminated card, my handwriting...isn't the best for "at-a-glance reference".

**Possible Solutions**
There are two possible solutions here:
* Printing one keyboard with each key having the functions from all layers color-coded.  This would look something similar to [this image](https://ksr-ugc.imgix.net/assets/028/401/993/c40c14863bb7c9ca934cab491dc7e91d_original.png?ixlib=rb-2.1.0&w=680&fit=max&v=1583955112&auto=format&frame=1&lossless=true&s=9ba9809a182ad89a0025ab18fc47f069) on [the Keyboardio Atreus product page](https://shop.keyboard.io/collections/keyboardio-atreus/products/keyboardio-atreus).  Bonus points if you can chose the color for each layer.
* Printing each layer as its own keyboard layout.  This would ideally look similar to [this image](https://ksr-ugc.imgix.net/assets/028/440/943/a0cc2331c9bf601b387ce86eebfb1fcd_original.png?ixlib=rb-2.1.0&w=680&fit=max&v=1584312206&auto=format&frame=1&lossless=true&s=87a9b1e3fc55034fd5dffa98c881165e), again from the Atreus product page.  And again, bonus points if you can chose the color for each layer.

And, of course, bonus points for being able to print both versions.

**Alternatives**
* I could always take screenshots of the (admittedly gorgeous) Chrysalis UI.  That makes for some great images.  But getting those printed - especially with more than one layer on the same page - presents some challenges.
* Another option is to reproduce the entire layout in the QMK Configurator, which does have a printing feature.  Have done this, but I hate printouts that don't give credit to the software actually making it happen.

clemensw, 2021-01-12:
Related to #583

obra, 2022-03-22:
(Closing in favor of #583)

closed 2022-03-22 by obra