keyboardio/Chrysalis #622: Add some CLI interface

eloyesp, 2020-12-30:
There are multiple issues related to Chrysalis export related functionality, but I think that the main issue is not just that, but interoperability, so Chrysalis cannot be extended without modification, and that is an anti-pattern.

I see Chrysalis doing multiple different things, and while it makes it a great experience at the begining and should be a priority, I think that extracting the core functionality out and making the interface interact with those parts could not only make it possible to extend it without modifying it, but also make it easier to maintain.

Currently it do load and dump configuration from the keyboard, store internal state as a json, upload firmware, import and export by layer with copy paste interface. All those actions could be extracted to a separate binary with this interface:

- chrysalis gui (start the GUI interface)
- chrysalis dump 1 (dump layout 1 from keyboard and returns a json)
- cat symbols.json | chrysalis load 1 (load symbols.json to the keyboard)
- chrysalis flash file

With that in place, export and import functionality can be scripted (#597,#592, #587), printing layouts (#621) can be done with external tools and the GUI functionality would not be affected on any way.

On the other hand, I'm not sure if I'm missing something, but moving from Chrysalis to a custom firmware is a bit difficult, may be an export current layer as custom firmware, could also be implemented as an external tool.

obra, 2020-12-30:
Chrysalis is intended as a graphical app and I don't think it makes sense
to make small commandline tools depend on it.

Most of the things you're looking for (other than flashing) are things you
can do from the commandline with the serial protocol spoken by Kaleidoscope
and Chrysalis (although we'd love help with additional docs and commandline
tools.)

These may be enough to get you started.

https://kaleidoscope.readthedocs.io/en/latest/plugins/Kaleidoscope-FocusSerial.html?highlight=FocusSerial
https://kaleidoscope.readthedocs.io/en/latest/plugins/Kaleidoscope-EEPROM-Keymap.html?highlight=focus#focus-commands

We've talked about exporting keymaps from Chrysalis in a format that can be
compiled into a sketch. It's something that is desirable, but not a huge
priority - most folks who use Chrysalis aren't interested in building their
own firmware and vice-versa. While we'd be thrilled to have the feature,
we're likely to focus on better ways to publish and share keyboard layouts
in Chrysalis and on the web first.

eloyesp, 2020-12-30:
No, I understand what is the idea of Focus, but the thing is that Chrisalis already talk using Focus (I have no tool to send serial messages and receive responses from the keyboard), and already manages a readable representation (that might change), but it does hide those features behind an UI, making it hard to extend.

While those links could me help to build the cli tools from scratch, it will just duplicate effort, it will be incompatible with Chrisalis when the json representation changes, it will not be compatible with tools made from other users that use a different representation, etc.

I feel like I cannot express myself clear, in part I'm typing on the atreus, and it is quite hard, I have it half configured and I'm suffering a lot, and while there are tools and toolchains, those seems not to be available to build upon, so most people build everything from scratch each time.

algernon, 2021-01-09:
As it happens, there's the `src/api/` structure, which used to be a separate repository, and we used to upload those modules to the NPM package archive - but maintaining that separately from Chrysalis turned out to be a huge burden and timesink, so they got merged, and the packaging stopped. We _could_ resume pushing those packages to npm, and then building CLI tools on top of them should be fairly easy, and since those tools would be using the same libs Chrysalis is using, they'd be compatible too.

I don't think building the CLI tools is a high priority for us right now, but publishing the API packages is something that'd make it a whole lot easier for others to build those tools. Does that sound like an acceptable compromise?

closed 2022-06-24 by algernon