keyboardio/Chrysalis #629: How to add new layers in v0.8.0?

kkga, 2021-01-13:
I feel like an idiot, because obviously I'm missing something, but where is the action to add a new layer in the latest release?

algernon, 2021-01-13:
It's not possible (never was) to add completely new layers - only as many layers are available as much the firmware supports. However, 0.8.0 will, by default, only show layers that aren't empty. There's a "Show empty layers..." button below the mapping table on the sidebar. If you click that, it will show the empty layers too, you can click any row to switch to that layer in Chrysalis, and start mapping keys.

Does this answer your question?

kkga, 2021-01-13:
Unfortunately, I don't see the "Show empty layers..." button. This is the full height of the sidebar (AppImage release).

[image]

algernon, 2021-01-13:
I see. That's a bug then, will fix in a bit. Thanks!

algernon, 2021-01-13:
Found the issue, pushed a fix, travis should build a binary in about 30 minutes or so. Sorry about that, I mixed up a condition, and it was completely backwards.

closed 2021-01-13 by algernon

kkga, 2021-01-17:
@algernon how do I get the updated binary that includes this fix? The releases page still lists only the 0.8.0 version.

YarmoM, 2021-01-18:
+1, hoping a new version is being released soon

algernon, 2021-01-18:
We now have development builds again, currently available here: https://github.com/keyboardio/Chrysalis/releases/tag/v0.8.1-snapshot

The README will be updated with the new links soon.

commit f522c21c (2021-01-13): Editor/Overview: Fix the condition for showing the more layers button

We want to show the "Show empty layers..." button when there are more layers,
but the condition was backwards, and this patch fixes that.

Fixes #629.