keyboardio/Chrysalis #630: Can't set key to Macro, default firmware Macro does not run.

mrvdb, 2021-01-15:
After upgrading chrysalis to 0.8.0 and reflashing my firmware with the 1.99.3 "board" version in Arduino my macro assignment stopped working.

Downgrading to 0.7.9 with the chrysalis supplied firmware helped restore the functionality.

In 0.8.0 the macros assignments are visible on the keys, but when pressing the 'Macros' button an empty window appears. 

[image]

No assignments related to macros can be made and existing macros, including the default macro in the default firmware, do not output anything.

Debug bundle is attached
[chrysalis-debug.bundle.zip](https://github.com/keyboardio/Chrysalis/files/5819358/chrysalis-debug.bundle.zip)

 - OS: arch linux
 - Chrysalis Version: 0.8.0

algernon, 2021-01-15:
I fixed the empty dialog on master, looking at the other parts now. Thanks for the detailed report!

mrvdb, 2021-01-26:
UI issue confirmed fixed, but macros still don't work. I get the feeling that is a separate issue from this one really, because I can't get my own macros to work at all anymore, even if I downgrade.

algernon, 2021-01-26:
Bug found and fixed on the Kaleidoscope side. I will be updating the firmware shipped with Chrysalis later today, and will likely cut a 0.8.2 release soon after.

algernon, 2021-01-27:
Firmware files are updated on master, so closing this. Fixed firmware will be in the 0.8.2 release.

closed 2021-01-27 by algernon