keyboardio/Chrysalis #633: Add all Windows layouts

necsoiu, 2021-01-18:
Please add all Windows layouts. 

For example  I'm using "Romanian (Programmers)".

Thank you very much.

algernon, 2021-01-19:
Is [this](https://unicode-org.github.io/cldr-staging/charts/latest/keyboards/layouts/ro.html#ro-t-k0-windows-extended) the layout you'd want?

The problem with enabling _all_ windows layouts is that it's a lot of layouts, making the already crowded dropdown even longer. We can do that, but I'd strongly prefer doing so while we improve the UX of the layout selection widget too.

algernon, 2021-01-19:
I _think_ I figured out a way to do this somewhat reasonably, but it requires a lot of massaging to make things appear consistent. The main problem is that the name of the various layouts in the CLDR database aren't consistent. For example, there's "United Kingdom", "English (India)", "United States-International", and "US". All of these are variations of an English layout, yet only one of them has "English" in its name. Out of all the `en-` layouts, only the India variant has english in it's name. There are similar issues with many other languages, too.

So what we can do, is have a mapping between the language code (`en`, etc) and the language name (`English`), and group layouts by filename, using the language name as a header. That still won't be perfect, because to find the "English UK" variant, you'll need to look for "United Kingdom" under the "English" header... oh! If we make the headers selectable too, then I guess that'll work...

closed 2021-01-19 by algernon

necsoiu, 2021-01-23:
> Is [this](https://unicode-org.github.io/cldr-staging/charts/latest/keyboards/layouts/ro.html#ro-t-k0-windows-extended) the layout you'd want?
>
> The problem with enabling _all_ windows layouts is that [...]

 YES.  This is it. :) I would be very grateful if you could add it.

PR #638 (2021-01-19): Editor: Improve the host layout selector
There are three major things done here: first, the layout selector was changed from a simple dropdown to an autocomplete-able input. Second, we now load _all_ layouts from CLDR, and third, we group them by language.

The result is, in my opinion, quite pleasing:

[image]

Turning the selector into an autocomplete-able input also has the advantage that when one starts to type in it, it takes the whole typed string into account.

Fixes #633.