keyboardio/Chrysalis #635: CI fine tuning

algernon, 2021-01-18:
With the change from Travis to GitHub Actions, a few things still need fine tuning. Rather than opening many tickets for each thing, I'll list them here in one. All of these should be fairly straightforward.

- [ ] Update the README to point to relevant places:
  - [ ] The development build should point to the latest pre-release, or at least to `releases`. Image should be: [image]
  - [ ] The release build should point to `releases/latest`, rather than `releases`, and the image should be updated to: [image]
- [ ] Start tagging releases as `v$VERSION`, rather than `chrysalis-$VERSION`. We want to sort production releases higher than (previous) pre-releases. Doublecheck if we need this, or if releases are time-based.
- [ ] Update the release scripts in `tools/release/` based on the above steps: if we hardcode the latest pre-release link, our scripts should update the README too.
- [ ] Drop unused files:
  - [ ] `tools/get-artifact`
  - [ ] `tools/s3-upload.js` and `tools/s3-upload.js.orig`
- [ ] Figure out macOS signing

algernon, 2021-01-19:
Releases are sorted by the tag, not date, so we need to tag releases and pre-releases the same way.

algernon, 2021-01-19:
Apart from macOS signing, this is all on master now. Keeping this open for the signing, though.

Rom1deTroyes, 2021-01-26:
Just a feedback, here :

It seems that Release build shows developpement snapshots as availlable update.

I've noticed this on the morning, but the 0.8.1 release seems to take the place over the snapshot version...
... May be a check to do on the next release process ?

algernon, 2021-01-27:
The version checking _should_ have been fixed in 0.8.1. If you have 0.8.1 installed, and it shows 0.8.2-snapshot as an upgrade, that's a bug. Not a CI bug, though, but a bug in our version checking. If that's the case, can you open a new issue about it, please?

Rom1deTroyes, 2021-01-28:
I will keep an eye for the >0.8.1 versions :-)
For reference, using Chrysalis 0.7.7 : 
[image]

algernon, 2021-01-28:
Yeah, that's kind of expected, and not something we can fix. We used to have _only_ production builds on GitHub as github releases, and we weren't using the pre-release variants, development builds were on S3. So Chrysalis versions prior to 0.8.1 were checking GitHub releases, and offered the latest one, even if that's a pre-release, because they were under the assumption that any release on GitHub is a prod build.

The only way to fix this properly would be to not use github pre-releases and go back to S3 for dev builds, at least until a grace period is over. I don't think that's worth the effort. Chrysalis 0.8.1+ should not offer snapshot builds as upgrades.

algernon, 2021-10-02:
Jesse implemented macOS signing, current snapshot builds are signed, and the next stable release will be, too. As such, this is now fully resolved.

closed 2021-10-02 by algernon