keyboardio/Chrysalis #636: Can't add a layer in 0.8.0 layout editor

jpjerkins, 2021-01-19:
**Description**
There appears to be no way to add a layer in the new layout editor

**What I Tried**
Steps to reproduce the behavior:
1. Set a key on Layer 0 to be a layer shift to a new layer that doesn't yet exist in the keymap
2. Push the layout to the keyboard (a Keyboardio Atreus, in my case)
3. There is no way to set keymaps for the new layer

**Expected behavior**
There should be some way to select the new layer.  Perhaps parsing all the layers looking for references to new layers should cause the new layer to appear in the table at the top-right of the keymap editor.

**Environment:**
 - OS: Windows 10 Pro v1909 build 18363.1316
 - Chrysalis Version: 0.8.0

jpjerkins, 2021-01-19:
It could be the problem is that the editor doesn't support more than 5 layers (IE, it only supports layers 0-4).  I was trying to add Layer 5.

I also tried exporting my layout, editing it by hand, then importing it back.  Turns-out all 10 layers get exported, with "unused" layers exporting as all transparent keys.  When I edited one of the keys on Layer 5, then imported back, the editor still only showed layers 0-4.  I swapped a couple keys on Layer 0 just to make sure the import was working (eliminating JSON formatting errors).  The keys I swapped on L0 were reflected in the editor after importing.

jpjerkins, 2021-01-19:
Attaching the hand-modified export that apparently imported fine with a key mapped on Layer 5, but that doesn't result in Layer 5 being shown in the 0.8.0 editor.  Note the Q and W keys swapped in the top-left of Layer 0, proving that Chrysalis is indeed importing this file successfully.

[Phil Atreus.Chrysalis.txt](https://github.com/keyboardio/Chrysalis/files/5837963/Phil.Atreus.Chrysalis.txt)

algernon, 2021-01-19:
There was a bug in the 0.8.0 release, it didn't show the "Show empty layers" button at the bottom of the mapping table. This has been fixed on master, and the fix is available in 0.8.1-snapshot development build: https://github.com/keyboardio/Chrysalis/releases/tag/v0.8.1-snapshot

(This is pretty much the same issue as #629

jpjerkins, 2021-01-20:
Issue is nicely fixed by 0.8.1-preview.  Closing the issue.  Thanks, @algernon!

closed 2021-01-20 by jpjerkins