keyboardio/Chrysalis #640: Key combos no longer intuitively displayed in Chrsalis 8.0/ 8.1

tmarzolf, 2021-01-22:
**Describe the bug**
A number of key combinations no longer are intuitively displayed in Chrysalis:
"# 592" (ALT+ Left arrow)
"# 591" (ALT+ Right arrow)
"# 331" (CTRL + PG UP)
"# 334" (CTRL + PG Down)
"# 4432"
"# 4139"
"# 4431"

**To Reproduce**
Steps to reproduce the behavior:
1. Open Chrysalis
2. Click on key to edit
3. enter Left Arrow + modifier as ALT
4. See error - "# 592" displayed 

**Expected behavior**
see A+ Left Arrow

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Win 10
 - Chrysalis Version: 8.1

**Additional context**
This issue also presented in 8.0, but was not an issue in 7.9

Rom1deTroyes, 2021-01-22:
Same behavior with my k.io Atreus for Chrysalis 0.8.1-snapshot under Debian GNU/Linux.
This happens also for chordings like <kbd>Shift</kbd>+<kbd>Enter</kbd>

I also noticed a strange display : when you add modifiers to a modifier key (eg. choose the left <kbd>Control</kbd> on the picker, and add the <kbd>Shift</kbd> modifier, you get `S+[Object][Object]`, except for <kbd>Alt</kbd> that is well displayed.

Here are screenshots from the same keys on Chrysalis 0.7.7 and Chrysalis 0.8.1-snapshot :
[image] [image]

algernon, 2021-01-22:
Ooops! I recently tweaked the keycode->label database, and it seems I forgot to update a few parts. Reproduced, fix coming soon, most likely tomorrow.

Thanks for the report!

tmarzolf, 2021-01-22:
Thank you!  

"# 582" (Alt + Prt Scrn) also displays incorrectly; adding this to the keycode -> label database would appear to at least partially close #529 (Make modifiers available to Prt Scrn... and other Misc characters)

commit 8ced27ed (2021-01-25): api/keymap/db: Add a lot of modifier-agumented codes to the db

We were missing the modifier-augmented code for Fx, navigation, numpad, spacing,
and a few other keys. This adds all of those to the database, so these keys can
be set up with Chrysalis once again.

Fixes #640 and #641.
closed 2021-01-25 by algernon