keyboardio/Chrysalis #641: 8.1 - unable to add multiple modifiers to arrow keys, PG up, PG down, end, home

tmarzolf, 2021-01-22:
**Describe the bug**
Chrysalis 8.1 no longer allows me to assign multiple modifiers to arrow keys,  PG up, PG down, home or end

**To Reproduce**
Steps to reproduce the behavior:
1. Select a key to edit
2. Select arrow key, PG up, PG down, end or home from "Pick a Key"
3. Add first modifier (e.g Control)
4. Error: All other modifiers are greyed out

**Expected behavior**
As with letter keys (and as in 7.9 and prior) multiple modifiers should be allowed

**Screenshots**
n/a

**Desktop (please complete the following information):**
 - OS: Win 10
 - Chrysalis Version: 8.1

**Additional context**
Ctrl+Shift+Arrow key "commonly" used in EXCEL

Rom1deTroyes, 2021-01-22:
I can confirm the same behavior with Chrysalis 0.8.1-snapshot under Gnu/Linux (Debian 10).

algernon, 2021-01-22:
Thanks for the report - reproduced, and I'm working on a fix. It's most likely coming tomorrow, I have a pretty good idea how to fix it nicely.

algernon, 2021-01-25:
I'm pushing a fix in a few moments. Turns out, this is exactly the same issue as #640.

closed 2021-01-25 by algernon