keyboardio/Chrysalis #642: Layer shift for next action can't reach layer 9

Rom1deTroyes, 2021-01-22:
**Describe the bug**

When trying to assign a `Layer shift for next action` to Layer 9, only layer 7 is reachable.

**To Reproduce**
Steps to reproduce the behavior:
1. Choose a blank key
2. Click on `Layer and keymaps`
3. Scroll down to `Move to Layer`
4. Increase the numbers : you can't go higher than 7.

(I got things on my Layer 9).

**Expected behavior**
Ability to Shift to layer 9 for the next action.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: GNU/Linux Debian10 
 - Chrysalis Version: Chrysalis 0.8.1-snapshot

**Additional context**

Rom1deTroyes, 2021-01-22:
Just seen the comment in the code :

```JS
class LayerKeysBase extends React.Component {
  getMaxLayer = () => {
    const { keymap, selectedKey, layer } = this.props;
    const key = keymap.custom[layer][selectedKey];
    let max = keymap.custom.length - 1;

    if (db.isInCategory(key.code, "layer") && key.categories[1] == "oneshot") {
      // for oneshots Kaleidoscope supports max 8 layers so index is 7
      return Math.min(max, 7);
```

Is it relevant to open an issue over Kaleidoscope to extend oneshots max layers to the numbers of layers embedded by default on the Atreus ?

algernon, 2021-01-25:
Oneshot only supports 8 layers for technical reasons, it's not really feasible to extend it further: if we did, we would either need to break backwards compatibility (not an option), or split the OSL ranges in two, which would add considerable complexity and code size.

I'm afraid we need to stick to 8 layers with OSL for the foreseeable future.

(Shift To Layer, Lock to Layer, and Move To Layer allow you to go up to layer 31, though. OneShot and DualUse are limited to 8 layer.)

obra, 2024-02-26:
This is now documented

closed 2024-02-26 by obra