keyboardio/Chrysalis #647: Croatian keyboard layout is not available for selection

isipka, 2021-01-28:
Croatian (Croatia) keyboard layout is not available for selection

Rom1deTroyes, 2021-01-28:
Hello @isipka !
Can you tell which version of Chrysalis are you running (the number on the left menu like Chrysalis-0.8.2), and what is your OS (GNU/Linux, MacOS, Microsoft Windows) and version ?

The files are present in `static/cldr/keyboards/android/` and `static/cldr/keyboards/osx`, but it seems that [Croatian layout](https://unicode-org.github.io/cldr-staging/charts/latest/keyboards/layouts/hr.html) is only defined for Android, ChromeOS and OSX. :thinking:

isipka, 2021-01-28:
sorry, I'm running Chrysalis-0.8.2 on Windows 10 20H2

algernon, 2021-01-28:
For a bit of context: we currently slurp up all the layouts CLDR provides for Windows. We did so, because we were assuming that it would cover all languages covered by CLDR. Looks like that's not the case. Sadly, slurping it all up and just filtering out the duplicates doesn't work, because the files we have available aren't 100% consistent. (For example, the basic US English layout for Windows and OSX differ in subtle ways)

I think that at this time, the most feasible thing to do is pick up any missing languages from the `osx` or `chromeos` set as they are reported.

closed 2021-01-28 by algernon

isipka, 2021-01-28:
So if I understood you correctly you will add Croatian layout in one of the next releases ?

commit d29ae081 (2021-01-28): api/keymap: Load Croatian layouts from the osx set

We've been loading up all the windows layouts, under the assumption that it
covers all languages available via CLDR. It turns out, that's not the case.
Because naming is not 100% consistent, we can't load all the osx layouts and
filter out the duplicates, so we will be adding any missing languages as they
are reported.

With this patch, we lay the foundations for that, and load the Croatian layouts
from the osx set.

Fixes #647.