keyboardio/Chrysalis #648: unable to assign a layer shift **secondary action** to **backspace** key on default layout

isipka, 2021-01-28:
unable to assign a layer shift **secondary action** to **backspace** key on default layout
when I try to assign secondary action by expanding a dropdown menu and click on **layer shift when held** the option automatically changes to **modifier when held** and selects **altgr** as modifier key

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
it should assign a secondary action to backspace a layer shift function

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10 20H2
 - Chrysalis Version: 0.8.2

Rom1deTroyes, 2021-01-28:
This problem has just been corrected a few hours ago by commit 8a50189bdc8ccee16baf7e72b630c8f4769c3efc , and the latest snapshot should have the fix now !
Can you try with [Chrysalys-0.8.3-snapshot](https://github.com/keyboardio/Chrysalis/releases/tag/v0.8.3-snapshotl) ?

isipka, 2021-01-28:
it's working in 0.8.3 
thank you

closed 2021-01-28 by isipka

amperecz, 2021-02-05:
I had the same problem with my atreus and 0.8.3 has solved it. 

however, it seems that the problem still persists on my Model 01.