keyboardio/Chrysalis #649: Host layout selection issues

algernon, 2021-01-28:
**Describe the bug**

The widget letting us select the host-side layout has a number of shortcomings: ordering is weird, the initial rendering is slow, and searching isn't very convenient either.

**To Reproduce**

Steps to reproduce the behavior:

1. Go to the Editor screen.
2. Click on the "Linux layout" (or "Windows", or "macOS") dropdown.
3. Observe it takes a long time to open.
4. Open the developer console, and observe that we have a warning about it too (`[Violation] 'click' handler took 1150ms`)
5. Scroll around, and observe that the language groups aren't sorted by their display name (`Chinese` is sorted after `English`), but their language code instead.
6. Also observe that Material-UI complains about having duplicate headers.

**Expected behavior**

The selector should open _much_ faster, and sort the languages by their display name.

isipka, 2021-05-02:
IF possible the selector to initaly show the installed / active layouts on the users system first, and only search through the whole list as a second step.

obra, 2021-05-02:
Sadly, I don’t believe operating systems offer us that information.

algernon, 2022-07-27:
We have made _some_ progress on this: we're pregenerating the data now, so opening is much, much faster. Other issues still persist.

This is pretty similar to #1010, and since this was sooner, I'll close that one.

algernon, 2022-07-27:
With #1018 and #1019, and the CLDR data pregeneration already implemented on master, this issue is largely addressed.

One thing we talked about with @obra, which has not been mentioned here yet, is that ideally, we'd want to load *both* osx and windows layouts, and - by default - offer only the set available for the current platform. Linux would default to Windows (the chromeos layouts are unsuitable for our use case), because in my experience, the two largely agree on layouts, but both of them seem to disagree with osx from time to time.

closed 2024-11-11 by obra