keyboardio/Chrysalis #651: 0.8.3-Snapshot: "\" with secondary action "Layer #2" difficult to select in Chrysalis

tmarzolf, 2021-02-01:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Open Chrysalis 0.8.3-Snapshot and connect to Atreus
2. Pick a key "\"
3. Select secondary action
4. Choose "Layer Shift when Held"
5.  ERROR 1: Modifier when held shows, with modifier AltGr defaulted
6. Choose "Layer Shift when Held" again
7. Click up arrow on Target Layer twice
8. ERROR 2: Target Layer 0 results instead of expected layer 2
9. ERROR 3: Result is "Layer #2/ Menu"

**Expected behavior**
Expected result is "Layer #2 / \"  (Layer 2 and forward slash)

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Win 10
 - Chrysalis Version: 0.8.3 - Snapshot)

Rom1deTroyes, 2021-02-01:
I have a hard (but short !) time figuring that you've selected <kbd>\\</kbd> because it's striped by Github (must double it), and not present on the French keyboards :disappointed: 

I cannot reproduce this neither with Chrysalis-0.8.2 Appimage release, nor Chrysalis-0.8.3-snapshot (both under GNU/Linux).

Chrysalis-0.8.2 :
[image]

Chrysalis 0.8.3-snapshot :
[image]

Can you export your layout to a file (Layout Sharing -> Export to file...), and seek if you have `51779` or `51831` somewhere ?

Can you also try to search `Layer #2` and report them here (you must catch Layer#2/z, Layer#2/Menu and Layer#2/\\ at least) ?

When I export the 2 keys (what you want and what you get), I have :

```js
{"code":51779,"baseCode":49,"label":{"hint":"Layer #2/","base":"\\"},"target":2,"rangeStart":51218,"categories":["layer","dualuse"]}
```

```js
{"code":51831,"baseCode":101,"label":{"hint":"Layer #2/","base":"Menu"},"target":2,"rangeStart":51218,"categories":["layer","dualuse"]}
```

tmarzolf, 2021-02-01:
Attached is the export file corresponding to the screenshot I provided.

I am able to find one instance of 51779 (note that I have a properly working version of this key next to ESC in my screenshot)
`{"code":51779,"baseCode":49,"label":{"hint":"Layer #2/","base":"\\"}`

I am able to find zero instances of 51831

Menu, and presumably the erroneous key appear to show up as follows
`{"code":53623,"baseCode":1893,"label":{"hint":"Layer #2/","base":"Menu"}`

[0.8.3-snapshot Feb 1a.json.txt](https://github.com/keyboardio/Chrysalis/files/5905300/0.8.3-snapshot.Feb.1a.json.txt)

tmarzolf, 2021-02-01:
Original comment updated to properly show forward slash and work around GitHub display of this character - sorry for the confusion!

tmarzolf, 2021-02-02:
Here are the three instances of "Layer #2" from the export file:

`{"code":51759,"baseCode":29,"label":{"hint":"Layer #2/","base":"z"}`

`{"code":53623,"baseCode":1893,"label":{"hint":"Layer #2/","base":"Menu"}`

`{"code":51779,"baseCode":49,"label":{"hint":"Layer #2/","base":"\\"}`

tmarzolf, 2021-02-02:
This appears partially resolved in a current 0.8.3 snapshot... The intended "Layer #2 / \" (Layer 2 and forward slash) displays in Chrysalis, however action when held is forward slash, not the intended layer shift.

[image]

algernon, 2021-02-15:
> however action when held is forward slash, not the intended layer shift.

Are you running the default, or the experimental firmware for the Atreus? The secondary actions only work with the experimental firmware.

(Yes, we desperately need a way to see which plugins are enabled on the firmware side, so Chrysalis will stop offering keys that the firmware doesn't support - or offer them with a warning.)

tmarzolf, 2021-02-16:
I am now running the experimental firmware and 06d6767 and the expected functionality works as intended - I shift to layer 2 when holding forward slash.  This is no longer an issue from my perspective.

closed 2022-05-17 by algernon