keyboardio/Chrysalis #652: Unable to configure a key as Enter+Layer shift

jacereda, 2021-02-02:
**Describe the bug**
On 0.8.2 I'm unable to configure the Enter key as a layer shift when held. Had to rollback to 0.7.9.

**To Reproduce**
On layer 0, tell the Enter key it should shift to layer 6. Looks like it messes modifiers somehow and modifiers mess the layer shift. 
**Desktop (please complete the following information):**
 - OS: Windos 10
 - Chrysalis Version: 0.8.2

tmarzolf, 2021-02-02:
I think I'm seeing the same behavior with space as a layer shift when held on 0.8.3 Snapshot

Rom1deTroyes, 2021-02-02:
Can't reproduce this on Debian GNU/Linux (Chrysalis 0.8.3-snapshot.AppImage).

[image]

This problem should be resolved by commit commit 50de8b7e573f413dd0c1add21726893abda85645 (api/keymap: Fix the dualuse database entries). Does your 0.8.3-snapshot older than Thu Jan 28 11:28:52 2021 +0100 ?

Rom1deTroyes, 2021-02-02:
@jacereda : The Chrysalis-0.8.2 release has some problems with this function that has been corrected in the dev snapshot available here : https://github.com/keyboardio/Chrysalis/releases/tag/v0.8.3-snapshot

tmarzolf, 2021-02-02:
This works for me on a current 0.8.3 snapshot on Win 10

amperecz, 2021-02-06:
I had same problem on 0.8.2. for atreus, the 0.8.3 has resolved it. however for model 01, the secondary action still doesn't work on both versions. (on backspace and space).

algernon, 2021-03-04:
@amperecz Apologies for the late reply, but on the Model01, are you running the experimental firmware, or the default one? The default doesn't support secondary actions at all.

Meanwhile, as the key can be selected in 0.8.3-snapshot, I'm closing this issue. If it's still a problem, please feel free to reopen.

closed 2021-03-04 by algernon