keyboardio/Chrysalis #653: Support latin characters on plain US keyboard layout.

jacereda, 2021-02-02:
Maybe I'm missing something and this is possible right now without resorting to Kaleidoscope... If so, I'd appreciate a note on how to do it.

I don't want to use US International layout because dead letters are a PITA when programming (quotes need to be followed by a space, for instance).

At least on Windows, ALT+numpad allows to input arbitrary single-byte characters even when using the normal (non-international) layout. For instance, you can press ALT+130 to get e+acute.

**Describe the solution you'd like**

I'd love to have a new category of keys (Arbitrary Latin) where you can specify the numeric character and have the keyboard simulate ALT+numpad code. I guess this would please lots of european users.

algernon, 2021-11-25:
The [Unicode plugin](https://kaleidoscope.readthedocs.io/en/latest/plugins/Kaleidoscope-Unicode.html) does almost the same thing, and combined with Macros, you can have arbitrary letters input with a single keypress.

It isn't without downsides, however: for example, the input method works in _some_ windows applications, but doesn't in others, and designing the macros to support `Shift` (including oneshot, possibly) isn't trivial, either. 

It's certainly something we should make easier. I was thinking, that macro editing support is coming to Chrysalis in the not too distant future, so if we had a macro step that'd execute another step if shift is active, and yet another if it isn't, would let us make proper arbitrary symbol entries from within Crysalis. Might even be able to build a little UI on top of that, like a macro library. That'd effectively cover the "Arbitrary Latin" category you suggested, and more.

jacereda, 2022-01-03:
For those seeking an immediate solution for this, I've resorted to use https://github.com/thomasfaingnaert/win-us-intl-altgr

obra, 2024-11-11:
Being able to do this directly is a limitation of the USB HID protocol.

closed 2024-11-11 by obra