keyboardio/Chrysalis #655: 0.8.3-Snapshot - PrSc (Print Screen) no longer available in Chrysalis

tmarzolf, 2021-02-02:
**Describe the bug**
Print Screen no longer found in Pick a Key graphical tool or menus

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Pick a Key
2. Print Screen is not found as a valid Standard Key - even though it is on a standard keyboard layout

**Expected behavior**
Print Screen able to be configured from Chrysalis either from Pick a Key or from a menu

**Screenshots**

[image]

**Desktop (please complete the following information):**
 - OS: Win 10
 - Chrysalis Version: 0.8.3 Snapshot

**Additional context**
Add any other context about the problem here.

tmarzolf, 2021-02-02:
Workaround is to use custom key code 70

`{"code":70,"label":{"base":{"full":"Print Screen","1u":"PrSc"}}}`

algernon, 2021-02-15:
Oops!

The major reason it isn't there is because the key picker was originally displayed at the bottom, rather than in a popup dialog, and we wanted to save vertical space (that's why the Fx keys are on the right side too, rather than on top like on a standard keyboard). At that iteration, we had another way to select Print Screen, and a couple of other keys, which we no longer do.

I'll rearrange the selector, and add back print screen too. Thanks!

algernon, 2021-02-15:
Just pushed an update to master:

[image]

commit 06d6767e (2021-02-15): Editor: Rearrange the 104-key picker for a more standard layout

When we originally made the 104-key picker, it was displayed at the bottom of
the screen, and we wanted to save vertical space. For that reason, we ended up
putting the Fx keys to the right, and we did not include a few other keys, that
were offered elsewhere. Later on, we moved the picker from the bottom into a
popup in the center.

Being in a popup in the center, we no longer need to conserve vertical space,
nor do we need to exclude certain keys (which we no longer offer elsewhere!). As
such, rearrange the layout to a more sta [... cut]
closed 2021-02-15 by algernon