keyboardio/Chrysalis #656: Allow mod+key to send any key + potential mods

mpenet, 2021-02-03:
**Is your feature request related to a problem? Please describe.**

I am used to use some unixy bindings like C-m (triggers return), C-h (triggers delete), C-i (tab), C-M-h (triggers ctrl-backspace, aka backward-killword). These are quite useful on an atreus for instance, they are all on the home row and they also work on any other keyboard. It works everywhere bit not in Chrome and some other apps that highjacks these keys. 
 
**Describe the solution you'd like**

Because of that I would like to have the Atreus fw emit the resulting key combos directly when I use these combos. I did read about the various plug-ins but I couldn't find alternatives. 

**Describe alternatives you've considered**

I used software based solutions to fix that, but it's a bit hairy/slow.

hnrklssn, 2021-02-04:
What you can do instead, although it takes a bit of manual effort, is to create a new layer with all of the resulting outputs that you want. So the `m` key is `Return`, `h` is `Delete` and so on. While the keys that don't have special bindings are just bound to their normal key + the Ctrl modifier. So `c` is bound to `C-c`. Then your "Ctrl" key in your base layer is just mapped to shift to your Ctrl-layer when held.

mpenet, 2021-02-05:
> What you can do instead, although it takes a bit of manual effort, is to create a new layer with all of the resulting outputs that you want. So the `m` key is `Return`, `h` is `Delete` and so on. [...]

That's actually a good idea! I wonder if it's problematic with apps that have commands where you must hold ctrl down tho.

mpenet, 2021-02-05:
actually this won't work, I need to be able to use ctrl with keys that are on another layer as well.

hnrklssn, 2021-02-05:
How about you make it a "Layer shift for next action" to the layer I mentioned, and then add ctrl as a secondary action when held? So you can do `C m` to emit `Return`, while `C-m` is just `C-m`. Then you also get your holding ctrl that you mentioned. If that's not important, however, maybe inverting the actions would be more ergonomic, so you don't have to hold both ctrl and your layer shift down at the same time as chording with another key.
Of course, in theory you could fix your issue by adding yet another layer, and having ctrl on your secondary layer shift to this new layer. But it quickly starts using up a lot of layers, so I don't recommend it.

mpenet, 2021-02-06:
I tried using secondary actions but it's hit and miss, either I type too fast or I am doing something wrong, but it gets in the way more often than not. I think I am better off solving my issue at xkb (yuk) level for now :(, but it's fore sure not futureproof.

mpenet, 2021-02-07:
I ended up using the excellent https://github.com/snyball/Hawck instead of going too deep into firmware changes.

closed 2021-02-11 by mpenet