keyboardio/Chrysalis #658: "Fix It" button disables keyboard

hnrklssn, 2021-02-04:
**Describe the bug**
I got the warning that Chrysalis no longer supports mixing hardcoded and custom layers, and ignored it as I thought Fix It would take me to the relevant part of the code on Github in case I wanted to fix support for it myself. Also because everything still worked like normal, since I'm not using hardcoded layers anyways. Then I pressed it because I was curious, and the banner disappeared and my keyboard's backlighting turned off. It no longer responded to keypresses. Unplugging it and plugging it back in helped.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to keyboard
2. Click "Fix it"

**Expected behavior**
I expected to be taken to a website, since there wasn't anything that needed to be fixed with my keyboard as far as I could tell. Or at least have a tooltip when hovering over the button, explaining more concretely what the vague "Fix it" means. Does it remove hardcoded layers and reflash? Does it clear the EEPROM layers? Reading the source code it should disable hardcoded layers, but they weren't in use in the first place. And the settings still list Layer 3 as the default layer when Layer 0 from the layout editor is in use (see #657), so nothing seems to have changed in Chrysalis either.

**Desktop:**
 - OS: Windows 10
 - Chrysalis Version: 0.8.3-snapshot
 - Keyboard: Model 01 with Chrysalis 0.8.3 experimental firmware
 
PS: now that you support multiple keyboards maybe you want to add which keyboard the user used to the issue template?

hnrklssn, 2021-02-05:
Oh and I think now afterwards the layout is not quite right anymore. The layer modifier keys' layer index used to take hardcoded layers into account AIUI, and those numbers didn't change in the layout. So now my "shift to layer 4" key shifts to an empty layer instead of my 2nd EEPROM layer. I may be wrong about the before and after behaviour here though as I have been messing around a bit in Chrysalis, but I think this would be the cause. The layer shifts used to work, at least, and now they don't.

hnrklssn, 2021-02-05:
Another observation that I think is related to this: layer 1 in EEPROM now gets the numlock background lighting. But it's not the same physical keys, instead it seems based on what key they're mapped to, e.g. any key mapped to a number lights up red. So some keys light up when in layer 1 even though they're black in Chrysalis. And the numlock key fades up and down in brightness even though it's not mapped to anything. I can still make other keys (ones usually not in the numlock layer) light up in this layer with Chrysalis though. If I assign another color to a number key on this layer, however, it does not override the red backlight.

algernon, 2021-02-15:
There's a number of related, but kinda different issues in play here, so I'll try to unpack it, and offer some explanations, and solutions I have in mind.

First of all, I agree that the "Fix it" button needs some improvement, it's not clear what it does, and what the consequences are. I plan on doing two things about it: first, while the warning is displayed, Chrysalis should not allow edits, to avoid confusion. Second, pressing the button should pop up a dialog first, explaining what Chrysalis will be doing, why, and what the consequences are. We should probably reword the warning and the button, too. Perhaps renaming the button to "More..." or "More details..." would be a good idea, since it will pop up a dialog, which then, could have a "Fix it" button, or a "Proceed" one, or something along those lines.

Proceeding should also set the default layer to 0, and this should be explained aswell.

As for layer shifts not working: that's kind of expected, the keymap has `ShiftToLayer(4)`, and Chrysalis does not remap those, it doesn't touch the keymap at all, because that may have unintended side effects. It would be nice if we'd offer an option to do that, though, but it should be opt-in.

> layer 1 in EEPROM now gets the numlock background lighting

Yeah, that's an unfortunate side effect of the NumPad plugin of the Model01. It's... not a very smart plugin. For now, the best solution is to compile your own firmware and disable that plugin.

PR #675 (2021-03-22): Show a prompt screen instead of an alert when encountering mixed layers
Chrysalis no longer supports mixing hardcoded and custom layers, because they were confusing. However, showing an alert below the toolbar, but otherwise letting the user to continue also led to surprising behaviour, such as disabling the keyboard in some cases when pressing the "Fix it" button.

To combat these issues, these changes turn the alert into a separate screen, where the case, and the suggested solution is better explained. This prevents the user from editing the keymap and wondering why it doesn't work the way i [... cut]
closed 2021-03-22 by algernon