keyboardio/Chrysalis #660: Sticky Shift does't work

necsoiu, 2021-02-07:
**Describe the bug**

Slecting a Sticky Shift does't work well. If you hold it down it work but if you are making just one click for next key press it does't work.

**Desktop (please complete the following information):**
 - OS: [Windows 10]
 - Chrysalis Version: [0.8.3-snapshot]

tmarzolf, 2021-02-08:
This works for me with Atreus on Win 10 and 0.8.3 Snapshot...

How long do you wait before pressing sticky shift and the intended shifted key? A single press of sticky shift times-out after a couple of seconds.

Does double-tapping the sticky shift correctly cause shift lock for you?

[image]

necsoiu, 2021-02-13:
> How long do you wait before pressing sticky shift and the intended shifted key? A single press of sticky shift times-out after a couple of seconds.

I tried different intervals and nothing seems to work.

> Does double-tapping the sticky shift correctly cause shift lock for you?

 YES. Double-tapping is workink well, and also works well when I hold down the sticky shift.

closed 2022-06-26 by necsoiu