keyboardio/Chrysalis #661: build from source gyp failure

chaseadam, 2021-02-12:
**Describe the bug**
Attempting to build and run from checkout directory. Attempted with `latest` and tag `v0.8.2`.

I believe this has to do with the build attempting to use `electron-gyp` vs `node-gyp`, but I honestly don't have enough knowledge to understand the difference and why there would be a difference in in invocation between `yarn` and manual (as seen below).

My guess is there may be something up with my environment. Any pointers are appreciated.

Failure is reported as:

```
    gyp info using node-gyp@5.1.0
    gyp info using node@14.15.4 | linux | x64
    gyp info find Python using Python version 3.9.1 found at "/usr/bin/python3"
    gyp info spawn /usr/bin/python3
    gyp info spawn args [
    gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
    gyp info spawn args   'binding.gyp',
    gyp info spawn args   '-f',
    gyp info spawn args   'make',
    gyp info spawn args   '-I',
    gyp info spawn args   '/home/achasen/git/Chrysalis/node_modules/usb/build/config.gypi',
    gyp info spawn args   '-I',
    gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
    gyp info spawn args   '-I',
    gyp info spawn args   '/home/achasen/.electron-gyp/11.1.1/include/node/common.gypi',
    gyp info spawn args   '-Dlibrary=shared_library',
    gyp info spawn args   '-Dvisibility=default',
    gyp info spawn args   '-Dnode_root_dir=/home/achasen/.electron-gyp/11.1.1',
    gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
    gyp info spawn args   '-Dnode_lib_file=/home/achasen/.electron-gyp/11.1.1/<(target_arch)/node.lib',
    gyp info spawn args   '-Dmodule_root_dir=/home/achasen/git/Chrysalis/node_modules/usb',
    gyp info spawn args   '-Dnode_engine=v8',
    gyp info spawn args   '--depth=.',
    gyp info spawn args   '--no-parallel',
    gyp info spawn args   '--generator-output',
    gyp info spawn args   'build',
    gyp info spawn args   '-Goutput_dir=.'
    gyp info spawn args ]
    gyp info spawn make
    gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
    In file included from /home/achasen/.electron-gyp/11.1.1/include/node/v8.h:30,
                     from ../src/node_usb.h:12,
                     from ../src/node_usb.cc:1:
...

    make: *** [usb_bindings.target.mk:126: Release/obj.target/usb_bindings/src/node_usb.o] Error 1
    gyp ERR! build error 
    gyp ERR! stack Error: `make` failed with exit code: 2
    gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:194:23)
    gyp ERR! stack     at ChildProcess.emit (events.js:315:20)
    gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:277:12)
    gyp ERR! System Linux 5.10.13-200.fc33.x86_64
    gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
    gyp ERR! cwd /home/achasen/git/Chrysalis/node_modules/usb
    gyp ERR! node -v v14.15.4
    gyp ERR! node-gyp -v v5.1.0
    gyp ERR! not ok 
    error Command failed with exit code 1.
 
                    command=/usr/bin/node /usr/local/lib/node_modules/yarn/bin/yarn.js run install
                    workingDir=/home/achasen/git/Chrysalis/node_modules/usb
```

Strange thing is that if I change to `/home/achasen/git/Chrysalis/node_modules/usb` and run `/usr/bin/node /usr/local/lib/node_modules/yarn/bin/yarn.js run install` the build appears to work

```
➜  usb git:(b673345) ✗ /usr/bin/node /usr/local/lib/node_modules/yarn/bin/yarn.js run install
yarn run v1.22.10
$ prebuild-install --verbose || node-gyp rebuild
prebuild-install info begin Prebuild-install version 5.3.6
prebuild-install info looking for cached prebuild @ /home/achasen/.npm/_prebuilds/f28b5a-usb-v1.6.2-node-v83-linux-x64.tar.gz
prebuild-install http request GET https://github.com/tessel/node-usb/releases/download/v1.6.2/usb-v1.6.2-node-v83-linux-x64.tar.gz
prebuild-install http 404 https://github.com/tessel/node-usb/releases/download/v1.6.2/usb-v1.6.2-node-v83-linux-x64.tar.gz
prebuild-install WARN install No prebuilt binaries found (target=14.15.4 runtime=node arch=x64 libc= platform=linux)
gyp info it worked if it ends with ok
gyp info using node-gyp@5.1.0
gyp info using node@14.15.4 | linux | x64
gyp info find Python using Python version 3.9.1 found at "/usr/bin/python3"
gyp info spawn /usr/bin/python3
gyp info spawn args [
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/gyp/gyp_main.py',
gyp info spawn args   'binding.gyp',
gyp info spawn args   '-f',
gyp info spawn args   'make',
gyp info spawn args   '-I',
gyp info spawn args   '/home/achasen/git/Chrysalis/node_modules/usb/build/config.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/usr/lib/node_modules/npm/node_modules/node-gyp/addon.gypi',
gyp info spawn args   '-I',
gyp info spawn args   '/home/achasen/.cache/node-gyp/14.15.4/include/node/common.gypi',
gyp info spawn args   '-Dlibrary=shared_library',
gyp info spawn args   '-Dvisibility=default',
gyp info spawn args   '-Dnode_root_dir=/home/achasen/.cache/node-gyp/14.15.4',
gyp info spawn args   '-Dnode_gyp_dir=/usr/lib/node_modules/npm/node_modules/node-gyp',
gyp info spawn args   '-Dnode_lib_file=/home/achasen/.cache/node-gyp/14.15.4/<(target_arch)/node.lib',
gyp info spawn args   '-Dmodule_root_dir=/home/achasen/git/Chrysalis/node_modules/usb',
gyp info spawn args   '-Dnode_engine=v8',
gyp info spawn args   '--depth=.',
gyp info spawn args   '--no-parallel',
gyp info spawn args   '--generator-output',
gyp info spawn args   'build',
gyp info spawn args   '-Goutput_dir=.'
gyp info spawn args ]
gyp info spawn make
gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
...
make: Leaving directory '/home/achasen/git/Chrysalis/node_modules/usb/build'
gyp info ok 
Done in 1.04s.
➜  usb git:(b673345) ✗ echo $?
0
```

**To Reproduce**
Steps to reproduce the behavior:
1. checkout
2. run `yarn`

**Expected behavior**
yarn completes build without error

**Desktop (please complete the following information):**
 - OS: fedora 33
 - node: v14.15.4

algernon, 2021-02-12:
Sadly I can't see the exact error, but I have a suspicion... Can you try building with `export CXXFLAGS=-std=c++14` before running `yarn`?

Rom1deTroyes, 2021-02-13:
I had the same problem last week and started a git bisect to figure what happened... (wip)

I can confirm that the master branch is working on Debian GNU/Linux when adding `export CXXFLAGS=-std=c++14` before `yarn && yarn start`.

algernon, 2021-02-13:
The need for `-std=c++14` came when we updated our dependencies, the first change that introduced the flag is 86b711539b6edbf7c41824513114a399e83afb0f. It seems like the newest `usb` version (1.6.4) sets the flag itself now, so I'll try upgrading and removing the flag in the near future.

chaseadam, 2021-02-14:
`CXXFLAGS=-std=c++14 yarn` did the trick! Thanks for the pointers. `yarn start` works now.

Aside: curious why it was required when running `yarn` in one place rather than directly in the `/usb` dir.

obra, 2021-02-15:
Should we be patching something to force this?

algernon, 2021-02-15:
We can upgrade `usb` to get rid of the problem. Other solutions are not practical (as it would involve forking `usb`, which we really do not want to do, especially when there's a fixed version out).

obra, 2021-02-15:
Let's do that, then.

>

closed 2021-02-15 by algernon

Rom1deTroyes, 2021-02-15:
Just to confirm that f0123c8 did the trick too (at least on GNU/Linux Debian10 :-)

commit f0123c83 (2021-02-15): Update usb to 1.6.5

Update our `usb` dependency to 1.6.5, which compiles with -std=c++14 set. Also
remove the `CXXFLAGS` environment variable from the GitHub workflows, as they
are no longer needed.

Fixes #661.